NVlabs/Deep_Object_Pose #360: Can I run this model using the pre-trained data sets using a low-end GPU (GTX1650 laptop)?

KennethEladistu, 2024-05-28:
Currently I want to try Deep Object Pose Estimation model with the given data sets. My laptop is only GTX1650 with 4gb ram  can it run this on my laptop's webcam using the pre-trained data?

TontonTremblay, 2024-05-28:
Yeah I think you should be able to run it, if I remember the models loaded take 2gb of ram. Did you try?

KennethEladistu, 2024-05-29:
> Yeah I think you should be able to run it, if I remember the models loaded take 2gb of ram. Did you try?

Yes but when I ran the model using "D:/Deep_Object_Pose/weights/pretrained_weights.pth" --data "D:/Deep_Object_Pose/sample_data" --object cracker this is what appeared:

[image]

TontonTremblay, 2024-05-29:
I am not sure, there is an error with the detections happening here, @nv-jeff in your rework have you seen this?

nv-jeff, 2024-05-29:
That is the result when you don't detect one or more cuboid vertices. If you examine the `points` array after it has been set from the `obj` data, you will see something like `[None, None, None, None, None, None, None, None, (349.0386957881619, 159.16052822031338)]`  

This looks like a bug to me, introduced from andrewyguo's changes.  I have pushed a fix as https://github.com/NVlabs/Deep_Object_Pose/commit/68534008dbc26b524272c00e6bef68b44d5e4b88

closed 2024-05-29 by nv-jeff

mintar, 2024-05-31:
As an aside: 

> My laptop is only GTX1650 with 4gb ram can it run this on my laptop's webcam using the pre-trained data?

You won't have any problems with that. On one of my systems, I'm running DOPE on a GTX 1050 Ti 4GB without problems, and that GPU is even older than yours.