NVlabs/Deep_Object_Pose #362: Step by step on how to run DOPE on low-end device

KennethEladistu, 2024-05-30:
Hello I have a question, can I run this model without using ROS/ROS2 and ubuntu, I just downloaded the trained weights from your Github and also downloaded the HOPE 3D models.

I made an environment in anaconda prompt, git cloned your repo and installed the needed requirements from requirements.txt. I loaded the downloaded trained weights in the file folder together  with the HOPE 3D models. May I ask the next step on how to run the trained weights on my laptop's camera for testing? I'm new to deep learning and just wanna make sure my progress is right thank you!

TontonTremblay, 2024-05-30:
https://github.com/NVlabs/Deep_Object_Pose/tree/master/inference You can follow the steps here, you can run without ros.

KennethEladistu, 2024-06-04:
> https://github.com/NVlabs/Deep_Object_Pose/tree/master/inference You can follow the steps here, you can run without ros.

[image]

hello I have run the 

python inference.py --weights D:/Deep_Object_Pose/weights/pretrained_weights.pth --data D:/Deep_Object_Pose/sample_data --object cracker command and this is the output. Am I in the right track? May I ask what is the next step if I want to run it on my laptop's camera without using ROS, do you have any sample for the Configuration Files/projection matrix that I can use to run it on my laptop? thank you for the help.

TontonTremblay, 2024-06-04:
This looks like it ran nicely, do you see an output? I dont remember the interface to the inference script. You are currently running without ROS. So you should be able to replicate this on your laptop.

KennethEladistu, 2024-06-04:
> This looks like it ran nicely, do you see an output? I don't remember the interface to the inference script. You are currently running without ROS. So you should be able to replicate this on your laptop.

Hello, no there is no output, I just run the python inference.py --weights D:/Deep_Object_Pose/weights/pretrained_weights.pth --data D:/Deep_Object_Pose/sample_data --object cracker
Found 10 weights. on my Anaconda Prompt and it loaded the 10 models and after that no output, my laptop's camera did not open or any output at all. Thank you.

TontonTremblay, 2024-06-04:
I think you need to tell it to open the webcam. Check the arguments you can pass in the script.

KennethEladistu, 2024-06-04:
> I think you need to tell it to open the webcam. Check the arguments you can pass in the script.

oh okay okay is this for the inference.py script?

KennethEladistu, 2024-06-04:
> I think you need to tell it to open the webcam. Check the arguments you can pass in the script.

[image]
Hello, currently I modified my inference.py code and I have laptop camera working and the models are loading, but when I tried to  show my camera the Cheez-IT picture it does not detect anything, may I ask if I am doing the right thing for testing? thank you.

TontonTremblay, 2024-06-04:
yeah I dont think you loaded the weights for the cracker, you dont need to load objects you are not going to detect, it just slows down the process. I think you have the bbq sauce loaded that I can see, https://www.ebay.com/itm/124572623487 try these images.

KennethEladistu, 2024-06-04:
> yeah I dont think you loaded the weights for the cracker, you dont need to load objects you are not going to detect, it just slows down the process. I think you have the bbq sauce loaded that I can see, https://www.ebay.com/itm/124572623487 try these images.

[image]

this is my current result on my Iphone are the results bad for testing?

TontonTremblay, 2024-06-04:
@nv-jeff I think this is the same problem that your changes introduced?

KennethEladistu, 2024-06-06:
> @nv-jeff I think this is the same problem that your changes introduced?

good morning is there an update for this one? thank you

TontonTremblay, 2024-06-06:
We are at an NVIDIA even this week. I think Jeff told me he had a solution
but did not have time to push it because of the event. Early next week, it
should be there.

KennethEladistu, 2024-06-06:
> We are at an NVIDIA even this week. I think Jeff told me he had a solution but did not have time to push it because of the event. Early next week, it should be there.
> […](#)

thank you for the response!

nv-jeff, 2025-12-05:
What specifically is the problem you are seeing?  I don't know what changes Jon was talking about earlier, but there were some fixes pushed last year that may help.  

Are you using our pretrained weights, or is this something you trained yourself?  Are you certain the weights are for the object you are trying to detect?  What device are you running this on?

Secret-Lin, 2025-12-07:
Thank you for your response. I ran the code on Windows 10 after making a few minor modifications, and it successfully loaded the weights and executed correctly. 
However, I noticed that when the object occupies most of the image, it fails to be detected, whereas it can be detected when the object covers only a small region of the image. And I wonder How much occlusion will prevent an object from being detected? From my observation, maybe around 15%?

I am using a RealSense D435i camera, which provides depth images. I was wondering whether it is possible to use the depth information as an additional input.

I also noticed that there is a DOPE-Plus package. Since it does not provide pretrained weights, I am planning to download the dataset and train the model myself. Is DOPE-Plus significantly better than the original DOPE?