NVlabs/Deep_Object_Pose #364: Inference doesn`t show any output of images and json file with projected cuboid

AyeshaHabib, 2024-06-05:
Hello, I have been working on creating a dataset in nvisii as per the repo. I trained it on 5000 images and 100 epochs, the debug file shows the correct orientation of bounding box. After that, I configured config files for dimension, class ID and weights. When I tried running inference, it worked fine and gives output but the json file in output is empty and generated output pictures shows no bounding boxes. I tried changing the threshold parameters in config file too but it doesnt produce any result either. Other than that the DOPE-ROS inference also doesnt show any output. Can you please guide me on what to do about it?

TontonTremblay, 2024-06-05:
Could you share a few examples of the belief maps? I think the inference.py can save them as well. Did you look at the tensorboard when you trained? How does it look, is inference belief map matching with GT? Try to give me a few examples of your data as well, are they symmetries?

AyeshaHabib, 2024-06-07:
I am running the Git repo inference file, i tried running it. I figured out that there was dimension mismatch in config file and 3D model so I am getting out in inference with projected cuboid but still not able to see output in ROS-Dope

AyeshaHabib, 2024-06-07:
[image]

AyeshaHabib, 2024-06-07:
[image]

TontonTremblay, 2024-06-07:
can you try to match the ratio in the webcam to match your training data? Can you post your 3d model here as well?

AyeshaHabib, 2024-06-10:
how should I match ratio? and I tried tuning threshold in config files but the output is confusing. Can you clear the understanding of relation of threshold parameter in config file with the dataset inference

AyeshaHabib, 2024-06-10:
[image]

AngeloPrete, 2025-04-14:
Hi what did you mean when you said "I figured out that there was dimension mismatch in config file and 3D model"?