NVlabs/Deep_Object_Pose #365: Depth image same as segmentation

utsavrai, 2024-06-09:
Hi, I am trying to get depth map along with the segmentation but the data generated is not clear in which format it is. I assumed the depth map must have depth value for each pixel and thus it can be used as ground truth depth for the synthetic RGB images and its corresponding segmentation. 

But as you can see below both depth and segmentation .exr files have same output.

Please let me know how to achieve the metric depth using nvisii data generation pipeline.

RGB image
[image]

Segmentation Image
[image]

Depth Image
[image]

please find the attached link to the original files for you reference. [google drive](https://drive.google.com/drive/folders/120Z4oh8dQS-mUeN9P6tcebdAF4iKXUZW?usp=sharing)

mintar, 2024-06-10:
When you converted the images to PNG, the values were cropped, so everything looks black or white after conversion. This is why the EXR format was chosen in the first place: it allows a higher range of pixel values.

If you open the image in an EXR viewer such as https://viewer.openhdr.org/, you can see that the depth image actually has the depth information:

[image]

If you double-click a pixel, you can see the value. For example, near the center of the object, the value is 1.669, which is the depth in meters.

utsavrai, 2024-06-10:
Thank you very much for the explanation @mintar. I also had a doubt that depth information was present and following up on your explanation I also wrote a python script to visualise the depth. Once again thanks for clarifying my doubt. 

```python

import numpy as np
import matplotlib.pyplot as plt
import OpenEXR
import Imath

# Define the file path
file_path = '00000.depth.exr'

# Open the EXR file
exr_file = OpenEXR.InputFile(file_path)

# Get the header to extract image dimensions
header = exr_file.header()
dw = header['dataWindow']
width = dw.max.x - dw.min.x + 1
height = dw.max.y - dw.min.y + 1

# Define the format for reading the channels
FLOAT = Imath.PixelType(Imath.PixelType.FLOAT)

# Read the R channel (depth values)
r_channel_str = exr_file.channel('R', FLOAT)
r_channel = np.frombuffer(r_channel_str, dtype=np.float32).reshape(height, width)

# Clip the depth values to a reasonable range for visualization
clipped_depth_image = np.clip(r_channel, 0, 10)  # Adjust the range as needed

# Check the minimum and maximum values to ensure precision
print(f"Min depth: {clipped_depth_image.min()}, Max depth: {clipped_depth_image.max()}")

# Save the depth image as a float32 numpy array
depth_image_path = 'depth_image_float32.npy'
np.save(depth_image_path, r_channel)

# Create a visualization and save it as a PNG file
plt.figure(figsize=(10, 8))
plt.imshow(clipped_depth_image, cmap='gray')
plt.colorbar(label='Depth Value')
plt.title('Depth Visualization')
plt.xlabel('Width')
plt.ylabel('Height')
plt.savefig('depth_visualization.png')
plt.show()

# Output paths
print(f"Depth image saved at: {depth_image_path}")
print("Visualization saved as: depth_visualization.png")

```

closed 2024-06-10 by utsavrai