NVlabs/Deep_Object_Pose #367: Terrible results on inference

joansaurina, 2024-06-12:
Hey.
In order to undesrtand how DOPE works I did my first training with and YCB object which was already trained by the authors with the weights available.

I generated my data using blenderproc_data_gen:
[(https://github.com/NVlabs/Deep_Object_Pose/tree/master/data_generation/blenderproc_data_gen)]

I used this command:
`python ../Deep_Object_Pose/data_generation/blenderproc_data_gen/run_blenderproc_datagen.py --nb_runs 1 --nb_frames 1250 --path_single_obj /joansaurina_working_dir/DOPE/objects/YOLO/006_mustard_bottle/006_mustard_bottle.obj  --nb_objects 5 --distractors_folder /joansaurina_working_dir/DOPE/objects/distractors --nb_distractors 10 --backgrounds_folder /joansaurina_working_dir/DOPE/hdr_maps --outf /joansaurina_working_dir/DOPE/data --width 1920 --height 1080 --nb_workers 4 --run_id 0 --focal-length 1400 --scale 45`

With the 2000 hdri background you suggested me, 19 different distractors, 5 times my object on each generated frame.

I run this command 16 times to have 20.000 frames. 

Here you can see an example:

**Image:**

[image]

**JSON:**
```
{
    "camera_data": {
        "width": 1920,
        "height": 1080,
        "camera_look_at": {
            "at": [
                -0.0,
                1.0,
                -7.549790126404332e-08
            ],
            "eye": [
                -0.0,
                25.0,
                -0.0
            ],
            "up": [
                1.0,
                0.0,
                0.0
            ]
        },
        "intrinsics": {
            "fx": 1400.0,
            "fy": 1400.0,
            "cx": 0.0,
            "cy": 0.0
        }
    },
    "objects": [
        {
            "class": "YOLO",
            "name": "YOLO_000",
            "visibility": 12686,
            "projected_cuboid": [
                [
                    1108.5007070466388,
                    336.1911830647591
                ],
                [
                    1078.1812421446348,
                    255.0891076541963
                ],
                [
                    1070.65097473424,
                    260.3795655918342
                ],
                [
                    1102.3030211876526,
                    344.74086262955416
[... 619 lines cut ...]
Train Epoch: 60 [6400/21954 (29%)] 	Loss: 0.000000007277878 	Local Rank: 0
Train Epoch: 60 [9600/21954 (44%)] 	Loss: 0.000000006779598 	Local Rank: 0
Train Epoch: 60 [12800/21954 (58%)] 	Loss: 0.000000008119192 	Local Rank: 0
Train Epoch: 60 [16000/21954 (73%)] 	Loss: 0.000000025056913 	Local Rank: 0
Train Epoch: 60 [19200/21954 (87%)] 	Loss: 0.000000011838531 	Local Rank: 0

After that, I run inference:

`python ./Deep_Object_Pose/inference/inference.py --weights  ./weights/mustard_60.pth --data ./weights --object mustard --exts png --outf ./weights --config config_pose.yaml --camera camera_info.yaml`

And I get empty camera and non objects detected in my test images and in every image I try.

I check my belief maps and they are very bad:

[image]

In the original weights you provided I could see the following:

[image]

I would appreciate some help,

Thanks

Joan

mintar, 2024-06-12:
You specified `--object mustard`, but your json has `"class": "YOLO"`. Try `--object YOLO` instead.

In short, your dataset doesn't contain a single object of class `mustard`, that's why the training didn't work.

joansaurina, 2024-06-12:
Hey @mintar thanks your help.
After this debugging I can finally see some decent results but not the ones I could expect:

[image]

On the test images it only detects (2/4) objects (1 really hidden) and the points are not very accurate.

As for the training loss, it does no seem no get down:

```
Train Epoch: 2 [3200/21954 (15%)]       Loss: 0.002900564344600         Local Rank: 0
Train Epoch: 2 [6400/21954 (29%)]       Loss: 0.005925509147346         Local Rank: 0
Train Epoch: 2 [9600/21954 (44%)]       Loss: 0.004374170210212         Local Rank: 0
Train Epoch: 2 [12800/21954 (58%)]      Loss: 0.004807741846889         Local Rank: 0
Train Epoch: 2 [16000/21954 (73%)]      Loss: 0.004936175886542         Local Rank: 0
Train Epoch: 2 [19200/21954 (87%)]      Loss: 0.005042205099016         Local Rank: 0
Train Epoch: 3 [0/21954 (0%)]   Loss: 0.005912763066590         Local Rank: 0
Train Epoch: 3 [3200/21954 (15%)]       Loss: 0.007230440154672         Local Rank: 0
Train Epoch: 3 [6400/21954 (29%)]       Loss: 0.005241629201919         Local Rank: 0
Train Epoch: 3 [9600/21954 (44%)]       Loss: 0.003206571796909         Local Rank: 0
Train Epoch: 3 [12800/21954 (58%)]      Loss: 0.005741830915213         Local Rank: 0
Train Epoch: 3 [16000/21954 (73%)]      Loss: 0.004557413049042         Local Rank: 0
Train Epoch: 3 [19200/21954 (87%)]      Loss: 0.002986140083522         Local Rank: 0
Train Epoch: 4 [0/21954 (0%)]   Loss: 0.003827466396615         Local Rank: 0
Train Epoch: 4 [3200/21954 (15%)]       Loss: 0.003044116543606         Local Rank: 0
Train Epoch: 4 [6400/21954 (29%)]       Loss: 0.004852563608438         Local Rank: 0
Train Epoch: 4 [9600/21954 (44%)]       Loss: 0.005033444147557         Local Rank: 0
Train Epoch: 4 [12800/21954 (58%)]      Loss: 0.004395484458655         Local Rank: 0
Train Epoch: 4 [16000/21954 (73%)]      Loss: 0.003537489799783         Local Rank: 0
Train Epoch: 4 [19200/21954 (87%)]      Loss: 0.004236795473844         Local Rank: 0
Train Epoch: 5 [0/21954 (0%)]   Loss: 0.004057863261551         Local Rank: 0
Train Epoch: 5 [3200/21954 (15%)]       Loss: 0.006070987787098         Local Rank: 0
Train Epoch: 5 [6400/21954 (29%)]       Loss: 0.003305417951196         Local Rank: 0
Train Epoch: 5 [9600/21954 (44%)]       Loss: 0.006066087633371         Local Rank: 0
Train Epoch: 5 [12800/21954 (58%)]      Loss: 0.003837038995698         Local Rank: 0
Train Epoch: 5 [16000/21954 (73%)]      Loss: 0.005226328969002         Local Rank: 0
Train Epoch: 5 [19200/21954 (87%)]      Loss: 0.004632766358554         Local Rank: 0
Train Epoch: 6 [0/21954 (0%)]   Loss: 0.004065050743520         Local Rank: 0
Train Epoch: 6 [3200/21954 (15%)]       Loss: 0.005211489275098         Local Rank: 0
Train Epoch: 6 [6400/21954 (29%)]       Loss: 0.006589157972485         Local Rank: 0
Train Epoch: 6 [9600/21954 (44%)]       Loss: 0.004496794659644         Local Rank: 0
Train Epoch: 6 [12800/21954 (58%)]      Loss: 0.004006389062852         Local Rank: 0
Train Epoch: 6 [16000/21954 (73%)]      Loss: 0.005213705822825         Local Rank: 0
Train Epoch: 6 [19200/21954 (87%)]      Loss: 0.002471640473232         Local Rank: 0
Train Epoch: 7 [0/21954 (0%)]   Loss: 0.004492865875363         Local Rank: 0
Train Epoch: 7 [3200/21954 (15%)]       Loss: 0.004473573062569         Local Rank: 0
Train Epoch: 7 [6400/21954 (29%)]       Loss: 0.004640904255211         Local Rank: 0
Train Epoch: 7 [9600/21954 (44%)]       Loss: 0.003035692032427         Local Rank: 0
Train Epoch: 7 [12800/21954 (58%)]      Loss: 0.003556061536074         Local Rank: 0
Train Epoch: 7 [16000/21954 (73%)]      Loss: 0.005741178989410         Local Rank: 0
Train Epoch: 7 [19200/21954 (87%)]      Loss: 0.004475364461541         Local Rank: 0
Train Epoch: 8 [0/21954 (0%)]   Loss: 0.005986265838146         Local Rank: 0
Train Epoch: 8 [3200/21954 (15%)]       Loss: 0.003541904035956         Local Rank: 0
Train Epoch: 8 [6400/21954 (29%)]       Loss: 0.004665576852858         Local Rank: 0
Train Epoch: 8 [9600/21954 (44%)]       Loss: 0.004690160043538         Local Rank: 0
Train Epoch: 8 [12800/21954 (58%)]      Loss: 0.004745676647872         Local Rank: 0
Train Epoch: 8 [16000/21954 (73%)]      Loss: 0.004688395652920         Local Rank: 0
Train Epoch: 8 [19200/21954 (87%)]      Loss: 0.003886136692017         Local Rank: 0
Train Epoch: 9 [0/21954 (0%)]   Loss: 0.006378377787769         Local Rank: 0
Train Epoch: 9 [3200/21954 (15%)]       Loss: 0.003509570378810         Local Rank: 0
Train Epoch: 9 [6400/21954 (29%)]       Loss: 0.005827599205077         Local Rank: 0
Train Epoch: 9 [9600/21954 (44%)]       Loss: 0.004743957892060         Local Rank: 0
Train Epoch: 9 [12800/21954 (58%)]      Loss: 0.004661967977881         Local Rank: 0
Train Epoch: 9 [16000/21954 (73%)]      Loss: 0.002835759660229         Local Rank: 0
Train Epoch: 9 [19200/21954 (87%)]      Loss: 0.004931921139359         Local Rank: 0
Train Epoch: 10 [0/21954 (0%)]  Loss: 0.004626110196114         Local Rank: 0
Train Epoch: 10 [3200/21954 (15%)]      Loss: 0.003018319839612         Local Rank: 0
Train Epoch: 10 [6400/21954 (29%)]      Loss: 0.003086642362177         Local Rank: 0
Train Epoch: 10 [9600/21954 (44%)]      Loss: 0.002127424115315         Local Rank: 0
Train Epoch: 10 [12800/21954 (58%)]     Loss: 0.004808292724192         Local Rank: 0
[... 73 lines cut ...]
[... cut ...]
Train Epoch: 22 [19200/21954 (87%)]     Loss: 0.004340017680079         Local Rank: 0
Train Epoch: 23 [0/21954 (0%)]  Loss: 0.002840435598046         Local Rank: 0
Train Epoch: 23 [3200/21954 (15%)]      Loss: 0.003147333860397         Local Rank: 0
Train Epoch: 23 [6400/21954 (29%)]      Loss: 0.003390608355403         Local Rank: 0
Train Epoch: 23 [9600/21954 (44%)]      Loss: 0.003638479625806         Local Rank: 0
```

Is this the training behavior I should expect? 
What does this Local Rank mean?

Thanks,

Joan

mintar, 2024-06-13:
"Local Rank" is only relevant if you're training on multiple GPUs in parallel.

The loss doesn't look too good, looks like nothing much is happening.

What do the belief maps look like?

joansaurina, 2024-06-13:
Hey @mintar,

Ground truth look like this:
[image]

And guess:

[image]

Are they weird?

I got there from here in train.py:

[image]

Thanks,

Joan

mintar, 2024-06-14:
> Are they weird?

Yes. The images are normalized, so "all grey" means "all black" (there are no bright peaks in the image). There should be bright spots similar to the ground truth. In other words, training is still not working.

joansaurina, 2024-06-16:
I have chacked my data again using debug.py:
https://github.com/NVlabs/Deep_Object_Pose/blob/master/common/debug.py

My data looks like this:

[image]

[image]

Are the points in the order they should be?
Should the objects be bigger?
Does the model work better with less objects?

Also for mustard objects I am assuming that the measurements in https://github.com/NVlabs/Deep_Object_Pose/blob/master/config/config_pose.yaml are right. Should they change depending on the scale I give to the object on the syntethic data or it's only important for real size?

"mustard": [9.6024150848388672,19.130100250244141,5.824894905090332]

Thanks,

Joan

mintar, 2024-06-16:
There are multiple steps:

1. Training: ground truth belief maps → weights
2. Network inference: weights → estimated belief maps
3. solvePnP: estimated belief maps + object dimensions + camera intrinsics → object pose

In your case, we know that step 2 already produces garbage belief maps, so we don't need to worry about step 3 for now. The object dimensions, order of points and camera intrinsics are not necessary in step 2 yet. The only input so far are the ground truth belief maps, and they look fine. I would guess you still have not trained on the correct object. Did you retrain with `--object YOLO`?

joansaurina, 2024-06-16:
Hey.
Yes I have done it. 
Now the loss does not get down :-( .

Well I prefered changing the json instead of runing on YOLO so my .json not look like:

```
{
    "camera_data": {
        "width": 1920,
        "height": 1080,
        "camera_look_at": {
            "at": [
                -0.0,
                1.0,
                -7.549790126404332e-08
            ],
            "eye": [
                -0.0,
                25.0,
                -0.0
            ],
            "up": [
                1.0,
                0.0,
                0.0
            ]
        },
        "intrinsics": {
            "fx": 1400.0,
            "fy": 1400.0,
            "cx": 0.0,
            "cy": 0.0
        }
    },
    "objects": [
        {
            "class": "mustard",
            "name": "mustard_000",
            "visibility": 12914,
            "projected_cuboid": [
                [
                    750.9035681927193,
                    390.83822261522187
                ],
                [
                    761.1644572794883,
                    442.94067550197883
                ],
                [
                    599.8493668936467,
                    379.6746688876557
                ],
                [
                    579.875981257254,
                    324.84771093683037
                ],
                [
                    718.7154844479179,
                    444.08253438598456
                ],
                [
                    731.1318636098214,
                    494.0153711489393
                ],
                [
                    567.2783000747321,
                    428.3917933960647
                ],
                [
[... 165 lines cut ...]
                    560.0290647847079
                ],
                [
                    1088.1110407512347,
                    652.2617814972203
                ],
                [
                    1046.3811895920194,
                    616.3140102268816
                ],
                [
                    1105.8948850486595,
                    587.7758404873681
                ]
            ]
        }
    ]
}
```

What could I check to see if I have done it right?

Thanks

Joan

mintar, 2024-06-17:
> What could I check to see if I have done it right?

You could show your belief maps again and check whether they have clear peaks.

RenanMoreiraPinto, 2024-06-19:
Got the same problem the train dont look to work . I download the dataset for canned meat that NVSII provide for test and see if i get any changes.. they have a 60.000 imagens dataset  to test, did you try using train or train2 version?

mintar, 2024-06-20:
There have been quite a few changes to the repo recently, some of which broke stuff. Maybe it's worth checking out an older version.

joansaurina, 2024-06-20:
I am using the current `train ` version they have on this Github, not `train2`. @RenanMoreiraPinto 

Do you know one version that was working that was 100% working? @mintar 

Thanks

Joan

mintar, 2024-06-20:
> Do you know which version was 100% working well?

`scripts/train2/train.py` in a4fe3cd4b5a739defdba4818d0a183490237ad0b is working very well.

joansaurina, 2024-06-20:
Which version of torchvision and Pillow are you using? I am having trouble with dependencies @mintar

RenanMoreiraPinto, 2024-06-20:
Start testing with train2 .. i just had to change 2 things :

line 140 #parser.add_argument("--local_rank", type=int)  
to
parser.add_argument('--local_rank', '--local-rank', type=int, default=0) 
and add on line 155 local_rank = opt.local_rank

and line 259 transform.Scale  to  transform.Resize  ...

We shall see how it goes... Fingers crossed

joansaurina, 2024-06-20:
Same here, the loss does not seem to be getting down...

TontonTremblay, 2024-06-22:
@nv-jeff did you test your changes to make sure the training script worked correctly? Following this thread I would think they broke a few things. 

I would recommend using the dope repo before @nv-jeff changes, using nvisii to generate data and training with these scripts. Sorry about this to @joansaurina and @RenanMoreiraPinto

joansaurina, 2024-06-22:
Could you point us to the correct version? @TontonTremblay 

And yeah i'll try to change blenderoc for nvisii

TontonTremblay, 2024-06-22:
https://github.com/NVlabs/Deep_Object_Pose/tree/128631a23c827d2091cfd103c03c8c3a93fc6134

joansaurina, 2024-06-22:
It seems I cannot use nvisii due to my nvdia drivers...

TontonTremblay, 2024-06-22:
so sorry! what error are you getting?

RenanMoreiraPinto, 2024-06-22:
To use the nvsii I had to downgrade the nvidia drivers to 450 and use the
ubuntu 20.04, but didn't go well at middle of image creation the pc
shutdown..
I using only 2 gpu for train so take longer to get results.. testing the
train2 atm.   I using the shiny meat dataset..

RenanMoreiraPinto, 2024-06-24:
i got some results using the train2 with 20.000 photos and 60 epochs ...
[image]
[image]
[image]
will increase the dataset and raise the training time.

joansaurina, 2024-06-24:
Hey @RenanMoreiraPinto 

Are these results with data generated with nvsii and this version of the code https://github.com/NVlabs/Deep_Object_Pose/tree/128631a23c827d2091cfd103c03c8c3a93fc6134 ?

Could you describe how did you generate the data with nvsii? The drivers and version you mentioned?

And which objects are you training on? If you are interested we could share the dataset we create (I want 20 YCB objects)

Thanks;

Joan

joansaurina, 2024-06-25:
Hey @TontonTremblay 
I am having same problem you discussed here:

https://github.com/owl-project/NVISII/issues/167

Do you think with blendeproc with more images the training could go well?

Also the json on the meat can dataset you have available (done with nvsii) have more things than the json blenderroc generates: 

**Meat can json:**

```
{
    "camera_data": {
        "camera_look_at": {
            "at": [
                0,
                0,
                0
            ],
            "eye": [
                0,
                0,
                1
            ],
            "up": [
                1,
                0,
                0
            ]
        },
        "camera_view_matrix": [
            [
                5.960464477539063e-08,
                0.9999999403953552,
                0.0,
                0.0
            ],
            [
                -0.9999999403953552,
                5.960464477539063e-08,
                0.0,
                0.0
            ],
            [
                0.0,
                0.0,
                1.0,
                0.0
            ],
            [
                0.0,
                0.0,
                -1.0,
                1.0
            ]
        ],
        "height": 512,
        "intrinsics": {
            "cx": 256.0,
            "cy": 256.0,
            "fx": 618.038818359375,
            "fy": 618.038818359375
        },
        "location_world": [
            0.0,
            0.0,
            1.0
        ],
        "quaternion_world_xyzw": [
[... 238 lines cut ...]
                    772.8770394270964,
                    882.4040559918749
                ],
                [
                    693.5464390005867,
                    866.1487835534365
                ],
                [
                    741.3058754329253,
                    809.6437815881382
                ],
                [
                    759.7376446027137,
                    738.1167859921233
                ]
            ]
        },
        ...
        
```
Is it important to have local cuboid, location and location world?

Thanks

Joan

nv-jeff, 2024-06-25:
> Also the json on the meat can dataset you have available (done with nvsii) have more things than the json blenderroc generates

The older, NVisii data generation code produced a number of fields that were used for experiments and debugging. The newer blenderproc data generation code only emits the required fields with one exception:  it also generates the debugging/visualization values:  'location' and 'quaternion_xyzw', which describe the position and orientation of the object in the camera coordinate system.

The data generated for each object is described at lines 278-287 of `data_generation/blenderproc_data_gen/generate_training_data.py`

RenanMoreiraPinto, 2024-06-26:
> Hey @RenanMoreiraPinto
>
> Are these results with data generated with nvsii and this version of the code https://github.com/NVlabs/Deep_Object_Pose/tree/128631a23c827d2091cfd103c03c8c3a93fc6134 ? [...]

hi! I used the blenderproc version 20.000 imagens dataset and my own blender .obj of a door :p ......... i used the new train2 version i will try the old version i making imagens on nvsii but since i had to make a downgrade to the nvidia 450 i used a old computer... i will try the new version with 60.000 dataset.

RenanMoreiraPinto, 2024-06-26:
[image]
[image]
[00007.json](https://github.com/user-attachments/files/15992039/00007.json)
[image]
[00008.json](https://github.com/user-attachments/files/15992042/00008.json)
[image]
[00009.json](https://github.com/user-attachments/files/15992043/00009.json)
[image]
[00010.json](https://github.com/user-attachments/files/15992044/00010.json)
[image] i made a run file to run all to make it work i just add the obj file whit the textured version on the models path the good thing is use the interactive to see items and position 
OUT_FOLDER="output"
args=""
args+=" --scale 0.5"
args+=" --nb_frames 10000"
args+=" --nb_objects 2"
args+=" --nb_distractors 10"
args+=" --outf $OUT_FOLDER"
args+=" --width 500"
args+=" --height 500"

# args+=" --objs_folder models"
# args+=" --path_single_obj /home/renan/Deep_Object_Pose/data_generation/nvisii_data_gen/models/door/door.obj"

# args+=" --debug"
# args+=" --interactive"

python3 single_video_pybullet.py $args 

ubuntu 20.04

joansaurina, 2024-06-27:
I am finally getting decent results!

I think I had these problems:

1. I needed more data: I had 20,000 images and now I have 75,000.

2. The images I was using were not square: I was using 1920x1080, and this did not fit well into the network which uses 512x512. I generated my images now with size 512x512.

3. I was using the new script which must have something wrong. The one working is: https://github.com/NVlabs/Deep_Object_Pose/tree/128631a23c827d2091cfd103c03c8c3a93fc6134

I'll report when training is done with inference results, but my loss does look good now:

```
Train Epoch: 1 [0/81599 (0%)]   Loss: 0.067368142306805
Train Epoch: 1 [3200/81599 (4%)]        Loss: 0.041182152926922
Train Epoch: 1 [6400/81599 (8%)]        Loss: 0.039299402385950
Train Epoch: 1 [9600/81599 (12%)]       Loss: 0.039307989180088
Train Epoch: 1 [12800/81599 (16%)]      Loss: 0.038886558264494
Train Epoch: 1 [16000/81599 (20%)]      Loss: 0.040250405669212
Train Epoch: 1 [19200/81599 (24%)]      Loss: 0.038537904620171
Train Epoch: 1 [22400/81599 (27%)]      Loss: 0.039164066314697
Train Epoch: 1 [25600/81599 (31%)]      Loss: 0.038494884967804
Train Epoch: 1 [28800/81599 (35%)]      Loss: 0.039913102984428
Train Epoch: 1 [32000/81599 (39%)]      Loss: 0.037832036614418
Train Epoch: 1 [35200/81599 (43%)]      Loss: 0.037465162575245
Train Epoch: 1 [38400/81599 (47%)]      Loss: 0.038491453975439
Train Epoch: 1 [41600/81599 (51%)]      Loss: 0.037618502974510
Train Epoch: 1 [44800/81599 (55%)]      Loss: 0.038155931979418
Train Epoch: 1 [48000/81599 (59%)]      Loss: 0.036317259073257
Train Epoch: 1 [51200/81599 (63%)]      Loss: 0.037838213145733
Train Epoch: 1 [54400/81599 (67%)]      Loss: 0.035968508571386
Train Epoch: 1 [57600/81599 (71%)]      Loss: 0.035015475004911
Train Epoch: 1 [60800/81599 (75%)]      Loss: 0.035495396703482
 ...
Train Epoch: 5 [54400/81599 (67%)]      Loss: 0.022631796076894
Train Epoch: 5 [57600/81599 (71%)]      Loss: 0.021600604057312
Train Epoch: 5 [60800/81599 (75%)]      Loss: 0.021810073405504
Train Epoch: 5 [64000/81599 (78%)]      Loss: 0.021704383194447
Train Epoch: 5 [67200/81599 (82%)]      Loss: 0.023373922333121
Train Epoch: 5 [70400/81599 (86%)]      Loss: 0.021433854475617
Train Epoch: 5 [73600/81599 (90%)]      Loss: 0.022643189877272
Train Epoch: 5 [76800/81599 (94%)]      Loss: 0.022783258929849
```

Thanks;

Joan

phsilvarepo, 2024-07-01:
Hello there,

Just some quick question @joansaurina, in this new training you performed. Are you training using train.py or train2, in the version you are using? And are you also able to perform inference? There is not a lot of documentation in that version, so not sure if it is possible. 

Thanks for the help

nv-jeff, 2024-07-01:
I am investigating the differences in training with the older training code (in `train2`) and the newer code (in `train`).  So far, it's clear that the newer code trains much faster (more than twice as quickly), which was noted when it was introduced, but the results are poorer.  I will update this thread when the problem has been located and fixed.

joansaurina, 2024-07-03:
Hey @phsilvarepo I used train2. Yes I am performing inference! 

Feel free to ask mor questions

Joan

nv-jeff, 2024-07-14:
I've done code comparisons and several experiments and I believe I have found the source of the quality discrepancies between the original training code (`train2`) and the newer training code (`train`).  Ironically it was a bug in the older code that was causing the issue.

Specifically, the old dataloader --  in `train2/utils_dope.py` at [lines 185-187](https://github.com/NVlabs/Deep_Object_Pose/blob/master/train2/utils_dope.py#L185-L187) -- was loading the data in subfolders **twice**.  So, in effect, the original DOPE code was performing two epochs of training per reported epoch.  This bug would only affect users who put data in subfolders and not all in one flat directory.  This bug explains why, for the same number of reported epochs, the old code generated better results, as well as explaining why the new training code appeared to be twice as fast as the older code.

I have run comparisons of the fixed old code and the new code on a single machine with a single GPU,  having set the random seed to 1 and setting the number of worker threads to one in order to eliminate randomness in data loading. The loss decrease is identical to four decimal places (at the end of one epoch) and the results are equivalent.  There are some small differences (less than a pixel) in cuboid point location but I believe this is the result of slight differences in math resulting from refactoring.

I will make the code change to eliminate the double-loading issue, but will not close this issue for now.  If any of you (awesome! beautiful! thoughtful!) users find any continuing differences, please let us know immediately and I will work with you to determine the problem.

phsilvarepo, 2024-07-22:
Hello there,
Just a quick question regarding the inference, did the results improved @joansaurina? I am still struggling to achieve good results even after the @nv-jeff bug fix, as I continue to get incomplete cuboid defections in my inference.

joansaurina, 2024-08-03:
No. It does not work well. @phsilvarepo

czy1998916, 2025-04-19:
@joansaurina Hi ,I am training the Dope model, but I have noticed that some strange problems often occur randomly, so I would like to ask you for a copy of your train.exe. If you could send me a copy, I would be very grateful

closed 2025-06-16 by nv-jeff