NVlabs/Deep_Object_Pose #369: dope crashes in rviz

AyeshaHabib, 2024-06-24:
Hello, I am able to see points for cuboid on belied map but the dope gave me error as shown in screenshot attached herewith. It crashed immediately. Kindly let me know what can be done about it [
[image]
](url)

AyeshaHabib, 2024-06-24:
[image]

nv-jeff, 2024-06-25:
This was caused by an old bug in the ROS1 version of `detector.py`.  It has been fixed in change https://github.com/NVlabs/Deep_Object_Pose/commit/e85dadb4016a18e1f2a83eb3febb0c1296673561

AyeshaHabib, 2024-06-25:
worked, thank you so much

closed 2024-06-25 by AyeshaHabib