NVlabs/Deep_Object_Pose #371: Mix training data from different sources

joansaurina, 2024-07-01:
After training for the YCB object mustard, I have obtained results that are not very accurate. After checking the paper, I realized that you were combining two datasets: realistic and randomized.

I have now downloaded the FAT dataset.

Do you recommend training the mustard object of YCB by combining synthetic data from BlenderProc with:

Single images containing only the target object?

A mix of single and mixed images of all objects (so the model also learns not to detect anything)?

Thanks,

Joan

intelligencestreamlabs, 2024-07-01:
They have used two approaches for data generation (Blender proc and Nvisii), Nvisii needs linux, nividia drivers and gpus to use it.
I have read a tip, you can try to generate image with 5 times of your object and 10 distractors from google_scanned_models

joansaurina, 2024-07-01:
Sorry that is not what I'm asking there @intelligencestreamlabs