NVlabs/Deep_Object_Pose #374: Range of detection

AyeshaHabib, 2024-07-10:
Hello I have been working on the DOPE for the orange color module using webcam and realsense camera but the issue is for the range of detection. As soon as object gets in range of 50cm of camera, it stops detecting and only detects for the certain range. I tried calibrating camera extrinsic to work around it but it didn't worked either and still give the same result. The scale I used in configuration script is the dimension of actual object and its 3D cad model. I have attached the picture from NVISII data generated for reference. KIndly let me know how to improve range of detection
[image]

AyeshaHabib, 2024-07-10:
[image]

RenanMoreiraPinto, 2024-07-10:
hey you used ros1? do you have a docker version you using?  thanks!

TontonTremblay, 2024-07-10:
I think you will need renders in your training data where the object appears closer.

AyeshaHabib, 2024-07-11:
> hey you used ros1? do you have a docker version you using? thanks!
I used ros1 installed in my workstation already, haven't tried using docker version though.

AyeshaHabib, 2024-07-11:
@TontonTremblay  Can you please eloborate in it more, will be of great help. What exactly should I configure in single_video_pybullet for object to appear closer to camera in data generated using NVISII.