NVlabs/Deep_Object_Pose #378: Confusion about what training is currently working?

brumocas, 2024-08-09:
After reading some issues I am confused with what training from the current repository is actually working.
Should I use `train` or `train2`?

Also about the data generation can we use both `blenderproc` and `nvisii` or is there any other tool with better performance?
I noticed that `blenderproc` generates different .json files compared with `nvisii`. Is this a problem?

Kind Regards,
Bruno Costa

TontonTremblay, 2024-08-09:
@nv-jeff introduced most of the changes, I will let him answer, sorry about that.

nv-jeff, 2024-08-09:
As far as we can tell, both training codes work the same.  `train2` is the original code, used for the paper (plus a couple minor bugfixes).  `train` is a simplified version that may be more understandable to new users.  If you find a significant difference in training results, please let us know.

The datagen code that uses `blenderproc` and `nvisii` can be used on their own, or their data can be combined.  As with `train2`, the `nvisii` code is the original version used for the paper.  However, because the `nvisii` library is no longer supported for newer versions of Python, we have reimplemented the required functionality in the `blenderproc` code.

The JSON that the `nvisii` code generates contains a lot of debugging and no-longer-used experimental fields.  The JSON generated by the `blenderproc` code contains only necessary information for training and visualization.

brumocas, 2024-08-09:
I really appreciate the response, it will be very helpful.
Regards,
Bruno Costa

closed 2024-08-09 by nv-jeff