NVlabs/Deep_Object_Pose #380: Terrible Inference Results

brumocas, 2024-08-19:
I have been using DOPE to train the following 3D model for pose estimation:
[image]

### Dataset
I managed to create a training dataset with 70K images using `blenderproc` following the guidelines provided by some users in this repo.
Example of the dataset created for training:
[image]
[image]

```json
{
    "camera_data": {
        "width": 500,
        "height": 500,
        "camera_look_at": {
            "at": [
                -0.0,
                1.0,
                -7.549790126404332e-08
            ],
            "eye": [
                -0.0,
                25.0,
                -0.0
            ],
            "up": [
                1.0,
                0.0,
                0.0
            ]
        },
        "intrinsics": {
            "fx": 603.5535070631239,
            "fy": 603.5535070631239,
            "cx": 0.0,
            "cy": 0.0
        }
    },
    "objects": [
        {
            "class": "reinforced_bracket",
            "name": "reinforced_bracket_000",
            "visibility": 18575,
            "projected_cuboid": [
                [
                    297.6431993557023,
                    132.3477357092983
                ],
                [
                    380.6734967660226,
                    -40.56986347207226
                ],
                [
                    251.76245282443915,
                    -24.779491521610453
                ],
                [
                    202.01544894933795,
                    155.72150571904643
                ],
                [
                    265.29468703303786,
                    75.50055533982413
                ],
                [
                    331.5917556117397,
                    -109.0965649077263
                ],
                [
                    205.7943853609635,
[... 575 lines cut ...]
Train Epoch: 754 [64000/70000 (91%)]    Loss: 0.011239994317293         Local Rank: 0
Train Epoch: 755 [0/70000 (0%)]         Loss: 0.011620987206697         Local Rank: 0
Train Epoch: 755 [6400/70000 (9%)]      Loss: 0.010749894194305         Local Rank: 0
Train Epoch: 755 [12800/70000 (18%)]    Loss: 0.011371455155313         Local Rank: 0
Train Epoch: 755 [19200/70000 (27%)]    Loss: 0.011337259784341         Local Rank: 0
Train Epoch: 755 [25600/70000 (37%)]    Loss: 0.011619374155998         Local Rank: 0
Train Epoch: 755 [32000/70000 (46%)]    Loss: 0.011956928297877         Local Rank: 0
Train Epoch: 755 [38400/70000 (55%)]    Loss: 0.012012710794806         Local Rank: 0
Train Epoch: 755 [44800/70000 (64%)]    Loss: 0.012054434977472         Local Rank: 0
Train Epoch: 755 [51200/70000 (73%)]    Loss: 0.011906860396266         Local Rank: 0
Train Epoch: 755 [57600/70000 (82%)]    Loss: 0.011918991804123         Local Rank: 0
Train Epoch: 755 [64000/70000 (91%)]    Loss: 0.011677873320878         Local Rank: 0
Train Epoch: 756 [0/70000 (0%)]         Loss: 0.011554966680706         Local Rank: 0
Train Epoch: 756 [6400/70000 (9%)]      Loss: 0.010675063356757         Local Rank: 0
Train Epoch: 756 [12800/70000 (18%)]    Loss: 0.011981759220362         Local Rank: 0
Train Epoch: 756 [19200/70000 (27%)]    Loss: 0.011198280379176         Local Rank: 0
Train Epoch: 756 [25600/70000 (37%)]    Loss: 0.011707981117070         Local Rank: 0
Train Epoch: 756 [32000/70000 (46%)]    Loss: 0.011175634339452         Local Rank: 0
Train Epoch: 756 [38400/70000 (55%)]    Loss: 0.011116879060864         Local Rank: 0
Train Epoch: 756 [44800/70000 (64%)]    Loss: 0.012733404524624         Local Rank: 0
Train Epoch: 756 [51200/70000 (73%)]    Loss: 0.011903902515769         Local Rank: 0
```
I am using an NVIDIA RTX 4090 to train the model.

`Question2`: Should I keep training until I get a better loss or my approach is incorrect?

brumocas, 2024-08-20:
Thanks in advance for the help :muscle:
Take a look at my beliefs, what can you conclude from that?
I generated this using test data.

[image]
[image]
[image]

doruksonmez, 2024-08-20:
@brumocas, May I ask you how did you generate belief maps? Thanks.

brumocas, 2024-08-20:
> @brumocas, May I ask you how did you generate belief maps? Thanks.

Sure, I used the `inference.py`from `train2` folder with the additional command `--showbelief`.
The script had some issues that I had to solve.
See my forked repository with the working code https://github.com/brumocas/DOPE/tree/main/train2

doruksonmez, 2024-08-20:
That was why I asked since I'm also having problems with it. I will give your fix a try and let you know. Thank you so much.

TontonTremblay, 2024-08-20:
sorry about the problems, feel free to MR them. It looks like your model is working well, at least to my eye, I would lower the confidence a bit to see if you get a cuboid. But you should be getting raw points for sure with what I am seeing.

brumocas, 2024-08-21:
@TontonTremblay  I changed the confidence in the `ìnference.py` `line 112` from `train2` folder and got slightly better results.
 ```python
        #self.config_detect.threshold = 0.5
        self.config_detect.threshold = 0.1
```

Some inference samples:
[image]
[image]
[image]
[image]

Yet a lot of simple detections are missed when the beliefs are there:
[image]
[image]

[image]
[image]

This is my config for the inference:
```python
topic_camera: "/dope/webcam/image_raw"
topic_camera_info: "/dope/webcam/camera_info"
topic_publishing: "dope"
input_is_rectified: True   # Whether the input image is rectified (strongly suggested!)
downscale_height: 400      # if the input image is larger than this, scale it down to this pixel height

# Comment any of these lines to prevent detection / pose estimation of that object
weights: {
    "reinforced_bracket":"../myscripts/weights/net_rb_70k_4.pth",
}

# Type of neural network architecture
architectures: {
    'reinforced_bracket':"dope",  
}

# Cuboid dimension in cm x,y,z
# From meshlab Cuboid dimension in cm y,z,x
dimensions: {
    #"reinforced_bracket": [6.7643, 4.8871, 12.8142],
    "reinforced_bracket": [12.8142, 6.7643, 4.8871]    
}

class_ids: {
    "reinforced_bracket": 0,
}

draw_colors: {
    "reinforced_bracket": [255, 0, 0],  # red
}

# optional: If the specified meshes are not in meters, provide a scale here (e.g. if the mesh is in centimeters, scale should be 0.01). default scale: 1.0.
mesh_scales: {
    "reinforced_bracket":  0.01,
}

overlay_belief_images: True   # Whether to overlay the input image on the belief images published on /dope/belief_[obj_name]

# Config params for DOPE
thresh_angle: 0.5
thresh_map: 0.0001
sigma: 3
thresh_points: 0.1
```

Any suggestion on how to improve this model?
Thanks in advance :muscle:

TontonTremblay, 2024-08-21:
This looks like it, to me this looks good! I think possibly having data where the object falls onto a table with other objects could help your model, but this is probably enough for you to play with the model. You could also add randomness on the object colors and material propreties, I think the nvisii script has that in, well it should not be hard if you are using nvisii. Not sure in the other scripts.

brumocas, 2024-08-21:
> This looks like it, to me this looks good! I think possibly having data where the object falls onto a table with other objects could help your model, but this is probably enough for you to play with [...]

I appreciate your help 💪, I will close the issue for now

closed 2024-08-21 by brumocas