NVlabs/Deep_Object_Pose #382: My training vs Paper training

brumocas, 2024-08-26:
After playing around with DOPE I decided to train the cracker 3D object to compare my training with the weights obtained from the training for the project paper.

`3D Object`
[image]

`Dataset`
I generated 83K images with `blenderproc`following the guidelines provided in this repo/issues.

`Training`
I have been training this object for a couple days and my loss looks like this:
```bash
Train Epoch: 29 [35000/83967 (42%)]     Loss: 0.008937404491007
Train Epoch: 29 [40000/83967 (48%)]     Loss: 0.007676803041250
Train Epoch: 29 [45000/83967 (54%)]     Loss: 0.007751216180623
Train Epoch: 29 [50000/83967 (60%)]     Loss: 0.007393791805953
Train Epoch: 29 [55000/83967 (65%)]     Loss: 0.007743947207928
Train Epoch: 29 [60000/83967 (71%)]     Loss: 0.007903006859124
Train Epoch: 29 [65000/83967 (77%)]     Loss: 0.008371086791158
Train Epoch: 29 [70000/83967 (83%)]     Loss: 0.008013324812055
Train Epoch: 29 [75000/83967 (89%)]     Loss: 0.007521728985012
Train Epoch: 29 [80000/83967 (95%)]     Loss: 0.007015910465270
Train Epoch: 30 [0/83967 (0%)]  Loss: 0.007150554563850
Train Epoch: 30 [5000/83967 (6%)]       Loss: 0.007973545230925
Train Epoch: 30 [10000/83967 (12%)]     Loss: 0.007326924707741
Train Epoch: 30 [15000/83967 (18%)]     Loss: 0.007029242813587
Train Epoch: 30 [20000/83967 (24%)]     Loss: 0.007687418721616
Train Epoch: 30 [25000/83967 (30%)]     Loss: 0.007528588175774
Train Epoch: 30 [30000/83967 (36%)]     Loss: 0.008784147910774
Train Epoch: 30 [35000/83967 (42%)]     Loss: 0.008270633406937
Train Epoch: 30 [40000/83967 (48%)]     Loss: 0.008088813163340
Train Epoch: 30 [45000/83967 (54%)]     Loss: 0.007582063321024
Train Epoch: 30 [50000/83967 (60%)]     Loss: 0.007671120576560
Train Epoch: 30 [55000/83967 (65%)]     Loss: 0.006739565171301
Train Epoch: 30 [60000/83967 (71%)]     Loss: 0.008357232436538
Train Epoch: 30 [65000/83967 (77%)]     Loss: 0.007158810272813
Train Epoch: 30 [70000/83967 (83%)]     Loss: 0.009133071638644
Train Epoch: 30 [75000/83967 (89%)]     Loss: 0.009372244589031
Train Epoch: 30 [80000/83967 (95%)]     Loss: 0.008244232274592
WARNING:tensorboardX.x2num:NaN or Inf found in input tensor.
Train Epoch: 31 [0/83967 (0%)]  Loss: 0.008372079581022
Train Epoch: 31 [5000/83967 (6%)]       Loss: 0.008861823007464
Train Epoch: 31 [10000/83967 (12%)]     Loss: 0.009239616803825
Train Epoch: 31 [15000/83967 (18%)]     Loss: 0.008081949315965
Train Epoch: 31 [20000/83967 (24%)]     Loss: 0.007703350391239
Train Epoch: 31 [25000/83967 (30%)]     Loss: 0.008221860043705
Train Epoch: 31 [30000/83967 (36%)]     Loss: 0.008797293528914
Train Epoch: 31 [35000/83967 (42%)]     Loss: 0.007885038852692
Train Epoch: 31 [40000/83967 (48%)]     Loss: 0.006942533887923
Train Epoch: 31 [45000/83967 (54%)]     Loss: 0.007813187316060
Train Epoch: 31 [50000/83967 (60%)]     Loss: 0.008053245022893
Train Epoch: 31 [55000/83967 (65%)]     Loss: 0.008084481582046
Train Epoch: 31 [60000/83967 (71%)]     Loss: 0.008321086876094
Train Epoch: 31 [65000/83967 (77%)]     Loss: 0.007608712650836
Train Epoch: 31 [70000/83967 (83%)]     Loss: 0.008138785138726
Train Epoch: 31 [75000/83967 (89%)]     Loss: 0.005781981162727
Train Epoch: 31 [80000/83967 (95%)]     Loss: 0.009008388966322
Train Epoch: 32 [0/83967 (0%)]  Loss: 0.007689560763538
Train Epoch: 32 [5000/83967 (6%)]       Loss: 0.008272772654891
Train Epoch: 32 [10000/83967 (12%)]     Loss: 0.007926281541586
Train Epoch: 32 [15000/83967 (18%)]     Loss: 0.009008924476802
Train Epoch: 32 [20000/83967 (24%)]     Loss: 0.007618052419275
Train Epoch: 32 [25000/83967 (30%)]     Loss: 0.008020709268749
Train Epoch: 32 [30000/83967 (36%)]     Loss: 0.007400152739137
Train Epoch: 32 [35000/83967 (42%)]     Loss: 0.007994290441275
Train Epoch: 32 [40000/83967 (48%)]     Loss: 0.006565771531314
Train Epoch: 32 [45000/83967 (54%)]     Loss: 0.008539704605937
Train Epoch: 32 [50000/83967 (60%)]     Loss: 0.006846275646240
Train Epoch: 32 [55000/83967 (65%)]     Loss: 0.008406339213252
Train Epoch: 32 [60000/83967 (71%)]     Loss: 0.010190168395638
Train Epoch: 32 [65000/83967 (77%)]     Loss: 0.007146433927119
[... 569 lines cut ...]
The loss seems to be stuck around the `0.007` value after approximately `200 epochs`. Is it still decreasing?
I am concerned because the results are far from the obtained in the paper.

Some inferences with my training:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Some beliefs with my training:
00000_belief:
[image]
00100_belief:
[image]
00200_belief:
[image]
00300_belief:
[image]

What can I do to improve this result, any suggestion?
Thanks in advance :muscle:

TontonTremblay, 2024-08-26:
Train further I would guess, the belief looks really good, but not your pnp results, can you visualize the raw points? I think your cuboid order is off @nv-jeff I think this is similar to the error the other person is having.

brumocas, 2024-08-26:
I am not familiar with the raw points, can you elaborate on how to get them?
However, I think you are referring to the points attributed to each cuboid vertex.

`Training Dataset orientation`
I have checked my training data and the debug looks like this:
[image]
[image]

Based on this generated data the `X` indicating the box orientation is in the `Nutrition Facts` side of the cracker 3D model
[image]

`Config file`
After that I edited my config file and saw that I had different results when using the training vs the paper weight.
In the end the config stayed like this:
```bash
# Cuboid dimension in cm x,y,z
dimensions: {
    # For my training
    "cracker": [7.179999828338623, 16.403600692749023, 21.343700408935547],
    # For paper weights
    #"cracker": [16.403600692749023, 21.343700408935547,7.179999828338623] 
}
```
I had to follow the orientation that my 3D object had in meshlab as you can see in my first meshlab screenshot available in this issue (top left corner [size is in meters]) . You guys probably had a different one in your 3D model.

`Inference results`
Some inference new inference results:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

I think that the output now is closer to your paper training weight, but yours is something else that I am not capable to achieve :(
Any suggestion in how to improve my results? How did you build your dataset?

`Future Work`
If I want to apply this to train a novel 3D object do you think that [Diff-DOPE](https://diffdope.github.io/) might be a great option to obtain the final pose?

TontonTremblay, 2024-08-27:
can you draw these https://github.com/NVlabs/Deep_Object_Pose/blob/master/ros1/dope/src/dope/inference/cuboid_pnp_solver.py#L101 

obj_2d_points

so I think when you call obj_2d_points and obj_3d_points; the points are not aligned, for example let say the top right front is at index 0 in obj_2d_points but index 1 in obj_3d_points, that will cause your issues. 

possibly might be caused by not using the right of these 2. Which now I do not remember which is which. https://github.com/NVlabs/Deep_Object_Pose/blob/master/ros1/dope/src/dope/inference/cuboid.py#L82 

@nv-jeff Maybe we should clear all of this so there is only one way to generate a set of cuboid points.

brumocas, 2024-08-29:
Hello again, and sorry for the delayed response.
I managed to solve the problem by changing the `cuboid_pnp_solver.py` and `cuboid.py` with the ones you suggested, only copy and paste.

`Inference results`
This is how my inference looks:
[00000.json](https://github.com/user-attachments/files/16801884/00000.json)
[image]
[00100.json](https://github.com/user-attachments/files/16801885/00100.json)
[image]
[00200.json](https://github.com/user-attachments/files/16801886/00200.json)
[image]
[00300.json](https://github.com/user-attachments/files/16801887/00300.json)
[image]
[00400.json](https://github.com/user-attachments/files/16801889/00400.json)
[image]
[00500.json](https://github.com/user-attachments/files/16801890/00500.json)
[image]
[00600.json](https://github.com/user-attachments/files/16801891/00600.json)
[image]
[00700.json](https://github.com/user-attachments/files/16801892/00700.json)
[image]
[00800.json](https://github.com/user-attachments/files/16801893/00800.json)
[image]
[00900.json](https://github.com/user-attachments/files/16801894/00900.json)
[image]
[01000.json](https://github.com/user-attachments/files/16801896/01000.json)
[image]
[01100.json](https://github.com/user-attachments/files/16801897/01100.json)
[image]
[01200.json](https://github.com/user-attachments/files/16801898/01200.json)
[image]
[01300.json](https://github.com/user-attachments/files/16801899/01300.json)
[image]
[01400.json](https://github.com/user-attachments/files/16801901/01400.json)
[image]
[01500.json](https://github.com/user-attachments/files/16801902/01500.json)
[image]
[01600.json](https://github.com/user-attachments/files/16801903/01600.json)
[image]
[01700.json](https://github.com/user-attachments/files/16801904/01700.json)
[image]
[01800.json](https://github.com/user-attachments/files/16801905/01800.json)
[image]
[01900.json](https://github.com/user-attachments/files/16801906/01900.json)
[image]
[02000.json](https://github.com/user-attachments/files/16801907/02000.json)
[image]
[02100.json](https://github.com/user-attachments/files/16801909/02100.json)
[image]

I think this is it for now, I will continue to play with DOPE 
Thanks :muscle:

TontonTremblay, 2024-08-29:
ahhh this looks amazing good job, sorry I will check with @nv-jeff to see if he could make the update.

nv-jeff, 2024-08-29:
Good sleuthing!  I will take a look at the code and refactor the changes mentioned above.