NVlabs/Deep_Object_Pose #386: How to test model accuracy

brumocas, 2024-09-26:
Hello again, I have been training DOPE on custom objects and I am  getting some results.
The model is the following:
[image]

The inference results are like the following images:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

I want to implement this in a real robot application but first i need to evaluate model accuracy
Thanks in advance :)
How can I generate a dataset like the one you did for the cracker box, with ground truths?
[image]
[image]
[image]
(...)

TontonTremblay, 2024-10-01:
I wrote a script that in nvisii renders the data on a table like this. https://github.com/TontonTremblay/nvisii_mvs?tab=readme-ov-file#falling-object-scene I think this is what I used, there might be some options to make sure the object is upright.

brumocas, 2024-10-04:
It worked for now :muscle: 
Really appreciate you help

TontonTremblay, 2024-10-04:
Cool show me some renders you did :P would love to see more, I like when people are using my broken code I wrote :P

brumocas, 2024-10-05:
Sure :)  
I managed to get the table and the object in the scene after playing around with the config.
The hdri background scene is not showing up for some reason, any idea to solve this?
Is the code only using the light from the hdri file?

[image]
[image]
[image]

brumocas, 2024-10-24:
Solved

closed 2024-10-24 by brumocas