NVlabs/Deep_Object_Pose #388: Incomplete cuboid detection (Real hardware detection)

Tianyi20, 2024-10-01:
Hi, today i transfer my output training pth.file to hardware detection. My object is the red mug in YCB benchmark. However, when i set everything done to do detection, the result is very poor that can not detecte any object. And once very luckly it detecte the mug, a error occurs: 

Incomplete cuboid detection.   
result from detection: [None, None, None, None, None, None, None, None, (351.5094017726776, 92.23743706236147)]
Skipping.

[image]

Furthermore, i would admit my training result is also very poor. Here I show my result for a several pictures:

[image]

[image]

[image]

[image]

Here is the command that generate data by nvisii_data_gen: 

$ **python single_video_pybullet.py --path_single_obj model/mug/textured.obj --scale 1 --nb_frames 10000 --nb_distractors 10 --nb_objects 25**

$ **python -m torch.distributed.launch --nproc_per_node=1 train.py --data /home/xiangtianyi/src/Deep_Object_Pose/data_generation/nvisii_data_gen/output/output_example --object obj --batchsize 10 --lr 0.001 --epochs 100**

Also I cannot assign correct  object class name in the output of data generation, it only appears to be "obj".

[image]

 
Could give me any suggestions on these problems? I would be great appreciate!

TontonTremblay, 2024-10-01:
ok wow yeah it is not looking too good, can you share some belief maps results? 

as for the class name, I think there is a split with the "." you might need to fiddle with this in the data generating script to get the right name.

Tianyi20, 2024-10-02:
[image]
[image]
[image]
[image]

Ah, it seems the result is pretty bad, sad。。

But do you have any suggestion ideas? I would appreciate your help!

Tianyi20, 2024-10-02:
> ok wow yeah it is not looking too good, can you share some belief maps results?
> 
> as for the class name, I think there is a split with the "." you might need to fiddle with this in the data generating script to get the right name.

[image]
[image]

[image]
[image]

It doesn't look well. Sad
Could you help me about it?

TontonTremblay, 2024-10-02:
It looks good to me, the points it found are well around the mug, the data you have, I think the mugs are getting too big in the image. Also the mug is symmetrical when you are not seeing the handle, but it does not look too problematic. Also I thought that mug was red. is it not?

Tianyi20, 2024-10-02:
> It looks good to me, the points it found are well around the mug, the data you have, I think the mugs are getting too big in the image. Also the mug is symmetrical when you are not seeing the handle, but it does not look too problematic. Also I thought that mug was red. is it not?

Thank you for your reply. But what do you mean the mug are getting too big in the image? Does it means I need to shrink the ' --scale 1" in video generation.py?Also the mug is in red color, does it have any problem needed to be fixed?

TontonTremblay, 2024-10-02:
I am sure if you generate a dataset with like 2-3 mugs and change the the plane locations to be further away, your weights would work just fine, did you try on a webcam?

Tianyi20, 2024-10-02:
> I am sure if you generate a dataset with like 2-3 mugs and change the the plane locations to be further away, your weights would work just fine, did you try on a webcam?

I will try to do this. I used to use realsesne DB435 camera but only use colour stream. It can detect sugar, catchup with the weights that you have trained very well. But it cannot even detect the mug in an entire cuboid. sad

Tianyi20, 2024-10-02:
> I am sure if you generate a dataset with like 2-3 mugs and change the the plane locations to be further away, your weights would work just fine, did you try on a webcam?

By the way, how to change the the plane locations to be further away？

TontonTremblay, 2024-10-02:
can you check the belief on the webcam? 

https://github.com/NVlabs/Deep_Object_Pose/blob/master/data_generation/nvisii_data_gen/single_video_pybullet.py#L252-L255 check the -2 I think, play with that.

Tianyi20, 2024-10-04:
hi, sorry for the delay response. But I still have problem about detecting my object, red mug. The dope can publish nothing about the dimension about the mug. However, i find the belief map is quite good.

[image]

Tianyi20, 2024-10-04:
Here is my temporary training result. And the training is till going.

[image]
[image]
[image]
[image]

Tianyi20, 2024-10-04:
But i can really well detect the sugar, is there something important to publish data that i forget to do?
[image]

Tianyi20, 2024-10-04:
I have also changed the texture.obj to let it make more sense. The detection output is OK. Here is an output. But why i can't detect anything when move to the hardware camera?

[image]

Tianyi20, 2024-10-04:
@TontonTremblay Hi, could you help me with that? I will be a lot of appreciate!

TontonTremblay, 2024-10-04:
yeah this looks good, I think the real world test you tried your object might be too close to the camera, did you try moving it out a little bit more? Also make sure you show it with the handle! I am not sure how to interpret the results on the real world mug, the beliefs are clearly not correct, when I have seen this in the past it was somewhat confusing the symmetries and or too close to the camera.

Tianyi20, 2024-10-04:
> yeah this looks good, I think the real world test you tried your object might be too close to the camera, did you try moving it out a little bit more? Also make sure you show it with the handle! I [...]

Thank you for your reply. But it nearly cannot output anything. Sad...

[image]

Tianyi20, 2024-10-04:
> > yeah this looks good, I think the real world test you tried your object might be too close to the camera, did you try moving it out a little bit more? Also make sure you show it with the handle! I am not sure how to interpret the results on the real world mug, the beliefs are clearly not correct, when I have seen this in the past it was somewhat confusing the symmetries and or too close to the camera.
> 
> Thank you for your reply. But it nearly cannot output anything. Sad...
> 
> ![Screenshot from 2024-10-05 06-13-29](https://private-user-images.githubusercontent.com/131247591/373819111-e49a96a2-f81b-443d-9d4c-235357d038fa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgwODAzNTMsIm5iZiI6MTcyODA4MDA1MywicGF0aCI6Ii8xMzEyNDc1OTEvMzczODE5MTExLWU0OWE5NmEyLWY4MWItNDQzZC05ZDRjLTIzNTM1N2QwMzhmYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwNFQyMjE0MTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMTNiM2U0Mjg2NTQxZWYxZWIwYTI0M2I1MGE1YzVhZWIzMGJjZjMwMjQ2OTFlMGI2MTIxZjdmMzViNTUxMjE5JlgtQW16LVN

[image]
pZ25lZEhlYWRlcnM9aG9zdCJ9._FYYmEjHgLauCBYbrz-UZQ4i5vKnKE4KD4jcFsXPL4Q)

TontonTremblay, 2024-10-04:
hummm I would think the model did not train long enough, you are seeing good results on the training data though?

Tianyi20, 2024-10-04:
> hummm I would think the model did not train long enough, you are seeing good results on the training data though?

Here is the output on the training data. By the Inference.py.

[image]
[image]
[image]

[image]
[image]

Tianyi20, 2024-10-04:
> hummm I would think the model did not train long enough, you are seeing good results on the training data though?

Sad, but i am really in hurry to use this detection output. I don't know why your weights to the sugar can work but my weights to the mug cannot work given that it can detect the mug in inference.py. But... Could you help me training this mug?  I can give you the YCB link to this mug's data...

TontonTremblay, 2024-10-05:
I would love to help more than just providing insights, but I am sorry, I cannot do much more than that. The results you got on the training data are very encouraging, maybe try with different light, maybe more?

Tianyi20, 2024-10-09:
@TontonTremblay Dear Tonton, i can say the problem is too less background pictures that are used for training. I find the inference can only works on the specific one background used for training. Maybe the data generation part should add more background pictures like your papers write that you use nearlly 10,000 background pictures for training. But really thanks to your help! And I'd like to say dope is an amazing work as long as you have enough computational source. HHH

closed 2024-10-09 by Tianyi20

TontonTremblay, 2024-10-09:
that makes sense, https://drive.google.com/file/d/1lp36MgTlS4OFaH0vdsTFhyGFJpQDY2YX/view?usp=sharing here are more background, I was under the impression you used more than that one background.

TontonTremblay, 2024-10-09:
I think there is 1000 backgrounds