NVlabs/Deep_Object_Pose #391: Symmetries after data generation

phsilvarepo, 2024-10-08:
Hello, 

I have generated a dataset to train DOPE in Issac Sim, but there does not seems to exist a method to deal with symmetries, how. can I change the model or edit the labels of the generated data to match the requirements to achieve good resullts?

TontonTremblay, 2024-10-08:
Check in this repo, there is information about how to deal with the generated data I think.

phsilvarepo, 2024-10-08:
Hello @TontonTremblay, 

What I have read so far, it only mentions how to deal with symmetries when generating the data, not after. I already have a dataset, but I cannot achieve good results due to the present symmetries, since pixel-wise, with rotation, the difference will be minimal. To deal with symmetries is the current method, is just to show certain faces during the data generation? Or the method will change the labels?

Thanks

TontonTremblay, 2024-10-08:
@mintar do you think he could use the script you wrote to directly modify his dataset? I think so right?