NVlabs/Deep_Object_Pose #393: is no output in .json/.png normal in inference?

zhaowangji, 2024-10-16:
i have no output in .json/.png(created by inference.py), but it does not report somethin like:Incomplete cuboid detection.result from detection: [None, None, None, None, None, None, None, None, (253.31389956007237, 183.16830538526844)] Skipping.

the output:
(.json)
{
  "camera_data": {},
  "objects": []
}
(.png)
no modification on the raw png, just the same

i used debug.py, and found that it works in some pictures(marked the object and showed its estimation)

closed 2024-10-21 by zhaowangji

AngeloPrete, 2025-04-10:
How due you solve it? I have the same problem.

TontonTremblay, 2025-04-11:
what is your set up?

AngeloPrete, 2025-04-11:
I took a picture of a box with my phone and tried to detect the 6D pose with inference.py