NVlabs/Deep_Object_Pose #396: Tools Recomendation ???

HappySamuel, 2025-01-07:
What kind of tools are recommended for preparing the `.json` files in `sample_data/` directory relevant to the `.png` / `.jpg` images before proceed to next stage of using `train.py` ?

I tried the `labelme`, but the .json file generated looks like below, it isn't same as the one show in the `sample_data/` directory ? 

> {
>   "version": "3.16.7",
>   "flags": {},
>   "shapes": [
>     {
>       "label": "flat_screwdriver",
>       "line_color": null,
>       "fill_color": null,
>       "points": [
>         [
>           2805.8688524590166,
>           678.9508196721312
>         ],
>         [
>           2810.786885245902,
>           756.0
>         ],
>         [
>           2800.9508196721313,
>           816.655737704918
>         ],
>         [
>           2796.032786885246,
>           1315.016393442623
>         ],
>         [
>           2835.377049180328,
>           1315.016393442623
>         ],
>         [
>           2884.5573770491806,
>           1365.8360655737706
>         ],
>         [
>           2856.688524590164,
>           1451.0819672131147
>         ],
>         [
>           2858.3278688524592,
>           1510.0983606557377
>         ],
>         [
>           2873.0819672131147,
>           1587.1475409836066
>         ],
>         [
>           2892.754098360656,
>           1693.704918032787
>         ],
>         [
>           2891.1147540983607,
>           1964.1967213114754
>         ],
>         [
>           2843.5737704918033,
>           2011.7377049180327
>         ],
>         [
>           2845.2131147540986,
>           2034.688524590164
>         ],
>         [
>           2805.8688524590166,
>           2039.6065573770493
>         ],
>         [
>           2746.8524590163934,
>           2034.688524590164
>         ],
>         [
>           2710.786885245902,
>           2011.7377049180327
>         ],
>         [
>           2715.7049180327867,
>           1996.983606557377
>         ],
>         [
>           2681.27868852459,
>           1947.8032786885246
>         ],
>         [
>           2691.1147540983607,
>           1674.032786885246
>         ],
>         [
>           2714.065573770492,
>           1583.8688524590164
>         ],
>         [
>           2727.1803278688526,
>           1518.295081967213
>         ],
>         [
>           2723.9016393442625,
>           1452.72131147541
>         ],
>         [
>           2696.032786885246,
>           1377.311475409836
>         ],
>         [
>           2735.377049180328,
>           1319.9344262295083
>         ],
>         [
>           2773.0819672131147,
>           1311.7377049180327
>         ],
>         [
>           2771.44262295082,
>           818.2950819672132
>         ],
>         [
>           2758.3278688524592,
>           762.5573770491803
>         ],
>         [
>           2768.1639344262294,
>           680.5901639344263
>         ]
>       ],
>       "shape_type": "polygon",
>       "flags": {}
>     }
>   ],
>   "lineColor": [
>     0,
>     255,
>     0,
>     128
>   ],
>   "fillColor": [
>     255,
>     0,
>     0,
>     128
>   ],
>   "imagePath": "N1_0.jpg",
>   "imageHeight": 2848,
>   "imageWidth": 4272
> }

TontonTremblay, 2025-01-07:
In the repo we have two solutions, one with blender and the other one with nvisii.

HappySamuel, 2025-01-08:
Hi @TontonTremblay 

Thanks for the info. I have toilet bowl images, if i want to detect it and get the pose of it when train with DOPE, which script of nvisii shall i use to prepare the dataset `.json` file ?

TontonTremblay, 2025-01-08:
Do you have a 3d model?

HappySamuel, 2025-01-08:
> Do you have a 3d model?

Nope, i don't have. Currently only have `.jpg` files

TontonTremblay, 2025-01-08:
our tools assume a 3d models. So without one I dont think I can help you. sorry.

HappySamuel, 2025-01-08:
OK, then i try to generate one. What are the file extensions for the 3d models shall i prepare ?

HappySamuel, 2025-01-09:
Hi @TontonTremblay 

I want to ask, if the model trained by dope, can it be use for centerpose as well ? Or any modifications need to be amend ?

TontonTremblay, 2025-01-09:
I think both models are trained on the same data structure.

TontonTremblay, 2025-01-09:
As for the file, obj should be ok.