NVlabs/Deep_Object_Pose #397: Objects from YCB Dataset Not Visible in Rendered Images Without `--scale 100` When Running `generate_training_data.py`?

strongBrave, 2025-01-18:
When using BlenderProc to generate synthetic images of objects from the YCB dataset, the objects do not appear in the rendered images. However, I found that adding the argument `--scale 100` resolves this issue, and the objects become visible in the generated images.Interestingly, when using object models from the HOPE dataset, the objects are rendered correctly without needing to adjust the scale (i.e., with `--scale 1`).
My question is how the argument `scale` affect the rendering process?  And why does setting `--scale` 100 make the YCB objects visible? Is there a difference in how the two datasets define object scales or units? 
```
./run_blenderproc_datagen.py --nb_runs 1 --nb_frames 10 --path_single_obj ../models/003_cracker_box/google_16k/textured.obj --nb_objects 6 --distractors_folder ../nvisii_data_gen/google_scanned_models/ --nb_distractors0 --backgrounds_folder ../dome_hdri_haven/
```

[image]

[image]

TontonTremblay, 2025-01-19:
some of the models are in meters and other in cm. This is baked into the 3d model. Scale end up multiplying the vertices of the model by that value.

strongBrave, 2025-01-20:
Thanks for your response! I think this aligns with what I had in mind. However, I am still a bit confused by the annotation regarding the `--scale` parameter, as shown in the image below. Since the `scale = 100` is appropriate for YCB-V dataset with cm unit in 3d model, I believe the correct explanation should be:

If the object scale is in meters, then `scale=1`.
If the object scale is in centimeters, then `scale=100`.
This would mean that if the scale is in centimeters, the value should be multiplied by 100, as opposed to using `scale=1`. Please let me know if this modified version works.

[image]

closed 2025-08-05 by nv-jeff