NVlabs/Deep_Object_Pose #403: nvisii_data_gen_problme

czy1998916, 2025-03-20:
Thank you for your contribution. I previously used Centerpose for pose estimation, including dataset creation, model training, and more. I want to learn Dope now, but I have some questions while using NVSII to create datasets.
1.I want to know the difference between centerpose and dope and their respective environments of adaptation
2.Differences in the single-video-pybullet.py script`parser.add_argument(
    '--objs_folder_distrators',
    default='/home/rx/NVISII/Deep_Object_Pose/data_generation/nvisii_data_gen/data/',
    help = "object to load folder"
)`and`parser.add_argument(
    '--path_single_obj',
    # default='data/ASICS_GELBlur33_20_GS_BlackWhiteSafety_Orange/meshes/model.obj',
    default=None,
    help='If you have a single obj file, path to the obj directly.'
)`According to my understanding, '- objs_folder-distrators' is an interference term used to increase the robustness of the dataset. However, I saw the pose information of the interference term in JSON. Will this information affect the training results during the training process? For example, interference items will eventually be identified
3.Every time I simulate, the model flies out of the field of view of the image. I adjusted the parameters below, but it didn't work. I want to know how to adjust it to make it appear completely in the picture`    toy_transform.set_scale(visii.vec3(scale))
    toy_transform.set_position(
        visii.vec3(
             random.uniform(0.1, 2),
             random.uniform(-1, 1),
             random.uniform(-1, 1),
            
        )
    )
    toy_transform.set_rotation(
        visii.quat(
             random.uniform(0, 1),
             random.uniform(0, 1),
             random.uniform(0, 1),
             random.uniform(0, 1),
           
        )
    )`
4.When I started the visualization process, I was not quite sure how to operate it. Is there a corresponding tutorial

[image]

We look forward to your reply @TontonTremblay

czy1998916, 2025-03-21:
@TontonTremblay

TontonTremblay, 2025-03-21:
You need to pass object id. Check the training script argument and make
sure it matches the object of interest. It will ignore the distractors. If
you don’t pass a name it loads everything by default.

czy1998916, 2025-03-21:
So, for training objects of interest, adjustments are made during the training phase, right?
Can you answer a few other questions.
Thank you very much

TontonTremblay, 2025-03-21:
For the other params it is a lot of trials and errors. I would recommend to
use blender instead of nvisii. I believe the scripts are more up to date
for blender. Also blender allows you to debug the scene. You can change the
translations uniform to be not as big. Also if you are using the pybullet
version you can change the plane sizes to have a smaller space. Use it with
gui enable so you can visualize the scene a little better. Check with
ChatGPT or Claude in how to do that.

czy1998916, 2025-03-28:
Sorry, I only saw your reply because I was busy with other projects. Do you have any tutorials on how to use Blender @TontonTremblay

TontonTremblay, 2025-03-28:
no worries, hope your other project was successful, please check the readme, it should have everything you need to know.

czy1998916, 2025-03-28:
For some reasons, I still chose NVISII when I adjusted the following code to make the object of interest move and transform within the camera's field of view. But there was no response. Can you tell me the reason`    toy_transform.set_scale(visii.vec3(scale))
    toy_transform.set_position(
        visii.vec3(
             random.uniform(0.1, 0.1),
             random.uniform(0.1, 0.1),
             random.uniform(0.1, 0.1),
            
        )
    )
    toy_transform.set_rotation(
        visii.quat(
             random.uniform(0.1, 0.1),
             random.uniform(0.1, 0.1),
             random.uniform(0.1, 0.1),
             random.uniform(0.1, 0.1),
           
        )
    )` @TontonTremblay

TontonTremblay, 2025-03-28:
I am not sure I understand the question, you seen to be on a good start to debug what is going on in your original post.

czy1998916, 2025-03-29:
Let me elaborate on my problem. I want to adjust the code below to fix the object in a fixed position in the image, preventing it from being captured. However, my modifications have no effect, as shown in the picture, where all objects still appear randomly in the image.
  
 `toy_transform.set_scale(visii.vec3(scale))
    toy_transform.set_position(
        visii.vec3(
             random.uniform(0.1, 0.1),
             random.uniform(0.1, 0.1),
             random.uniform(0.1, 0.1),
            
        )
    )
    toy_transform.set_rotation(
        visii.quat(
             random.uniform(0.1, 1),
             random.uniform(0.1, 1),
             random.uniform(0.1, 1),
             random.uniform(0.1, 1),
           
        )
    )`

[image]
[image]
[image]
[image]
[image]

TontonTremblay, 2025-03-29:
Ahhh it all makes sense, it is because it is using the pybullet physical simulator to move the object. I am sorry, you are going to have to by pass the pybullet step and manually integrate the movement or placement.

czy1998916, 2025-03-29:
oh~，no，This problem may sound complicated, could you elaborate on the steps to solve it in detail? Alternatively, you may have relevant tutorial links. If this problem can be solved, I would be very grateful to you

czy1998916, 2025-03-31:
@TontonTremblay

TontonTremblay, 2025-03-31:
I would say try with something like cursor AI, the script is not that complicated for an AI to figure out, aider AI is also pretty good.

czy1998916, 2025-03-31:
Are you saying that I implemented this feature by adjusting the code instead of other applications

TontonTremblay, 2025-03-31:
put the code in cursor AI, ask it what you want to achieve.

czy1998916, 2025-04-01:
[image]
Sometimes there may be issues with the images, is this due to your system

czy1998916, 2025-04-01:
Hello, after a series of debugging, I have finally trained a 2-round result. I am currently modifying the config.pose. yaml file. May I ask where to obtain dimensions and meshes? What is the purpose of filling in the obj file? Auxiliary estimation? model_transforms: What is the function of this parameter? @TontonTremblay

czy1998916, 2025-04-02:
bro，I really need your help. If you have time, please reply to me. Thank you very much @TontonTremblay

TontonTremblay, 2025-04-02:
I am not sure I follow the questions, so I was not sure what sort of answers you wanted.

czy1998916, 2025-04-02:
Then I will elaborate on my problem and show you my progress.
The problem is as follows: I am now preparing to use inference.Py to validate the model I have trained (the model has 10000 pieces of data and has only been trained twice, just to streamline the process, which will be optimized later), and then I need to modify the config.pose. yaml file, but there are a few parameters that I am not very clear about.
1.`dimensions: {
# "cracker": [16.403600692749023,21.343700408935547,7.179999828338623],
# "gelatin": [8.918299674987793, 7.311500072479248, 2.9983000755310059],
# "meat": [10.164673805236816,8.3542995452880859,5.7600898742675781],
# "mustard": [9.6024150848388672,19.130100250244141,5.824894905090332],
# "soup": [6.7659378051757813,10.185500144958496,6.771425724029541],
# "sugar": [9.267730712890625,17.625339508056641,4.5134143829345703],
# "bleach": [10.267730712890625,26.625339508056641,7.5134143829345703],
# "peg_hole": [12.6,3.9,12.6],
# 'cube_red':[5,5,5],
# 'pudding':[49.47199821472168, 29.923000335693359, 83.498001098632812],
'buerjia_0':[8.3555002212524414, 7.1121001243591309, 6.6055998802185059]
﻿}`
Where should I go to obtain this size parameter
2.`model_transforms: {
#    "cracker": [[ 0,  0,  1,  0],
#                [ 0, -1,  0,  0],
#                [ 1,  0,  0,  0],
#                [ 0,  0,  0,  1]]
}`
What does this transformation matrix mean, what is its function, and where can it be obtained
3.`meshes: {
#    "cracker": "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/003_cracker_box/google_16k/textured.obj",
#    "gelatin": "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/009_gelatin_box/google_16k/textured.obj",
#    "meat":    "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/010_potted_meat_can/google_16k/textured.obj",
#    "mustard": "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/006_mustard_bottle/google_16k/textured.obj",
#    "soup":    "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/005_tomato_soup_can/google_16k/textured.obj",
#    "sugar":   "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/004_sugar_box/google_16k/textured.obj",
#    "bleach":  "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/021_bleach_cleanser/google_16k/textured.obj",
}`
What is the purpose of adding a mesh file and is it to assist in estimation？

Below, I will provide you with the training data from trian.py and the result graph obtained from inference.cy

Regarding model training, I used a 4080s command to run `python train.py --network dope --epochs 2 --batchsize 10 --outf tmp/ --data /home/rx/Deep_Object_Pose/data_generation/nvisii_data_gen/output/output_example`

The terminal output is as follows
`NVISII) rx@rx:~/Deep_Object_Pose/train2$ python train.py --network dope --epochs 2 --batchsize 10 --outf tmp/ --data /home/rx/Deep_Object_Pose/data_generation/nvisii_data_gen/output/output_example
start: 09:29:37.935131
load data: ['/home/rx/Deep_Object_Pose/data_generation/nvisii_data_gen/output/output_example']
load data: 
training data: 1000 batches
load models
ready to train!
Train Epoch: 1 [0/10000 (0%)]   Loss: 0.088830776512623
Train Epoch: 1 [1000/10000 (10%)]       Loss: 0.060813341289759
Train Epoch: 1 [2000/10000 (20%)]       Loss: 0.062941178679466
Train Epoch: 1 [3000/10000 (30%)]       Loss: 0.064253278076649
Train Epoch: 1 [4000/10000 (40%)]       Loss: 0.061212349683046
Train Epoch: 1 [5000/10000 (50%)]       Loss: 0.063338302075863
Train Epoch: 1 [6000/10000 (60%)]       Loss: 0.060447327792645
Train Epoch: 1 [7000/10000 (70%)]       Loss: 0.063735432922840
Train Epoch: 1 [8000/10000 (80%)]       Loss: 0.059589948505163
Train Epoch: 1 [9000/10000 (90%)]       Loss: 0.060118705034256
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
Train Epoch: 2 [0/10000 (0%)]   Loss: 0.060208972543478
Train Epoch: 2 [1000/10000 (10%)]       Loss: 0.061098180711269
Train Epoch: 2 [2000/10000 (20%)]       Loss: 0.058770846575499
Train Epoch: 2 [3000/10000 (30%)]       Loss: 0.060757033526897
Train Epoch: 2 [4000/10000 (40%)]       Loss: 0.061982102692127
Train Epoch: 2 [5000/10000 (50%)]       Loss: 0.060894373804331
[... 414 lines cut ...]
#    "soup":    "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/005_tomato_soup_can/google_16k/textured.obj",
#    "sugar":   "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/004_sugar_box/google_16k/textured.obj",
#    "bleach":  "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/021_bleach_cleanser/google_16k/textured.obj",
}

# optional: If the specified meshes are not in meters, provide a scale here (e.g. if the mesh is in centimeters, scale should be 0.01). default scale: 1.0.
mesh_scales: {
    # "cracker": 0.01,
    # "gelatin": 0.01,
    # "meat":    0.01,
    # "mustard": 0.01,
    # "soup":    0.01,
    # "sugar":   0.01,
    # "bleach":  0.01,
    "buerjia_0": 1.0
}

# Config params for DOPE
thresh_angle: 0.5
thresh_map: 0.0001
sigma: 3
thresh_points: 0.1
`

Sorry, I have a bit too much to ask about. Thank you very much for your efforts.

czy1998916, 2025-04-02:
@TontonTremblay Sorry, I have a bit too much to ask about. Thank you very much for your efforts.

TontonTremblay, 2025-04-02:
I think things are clear on the readme. The size is for the PNP. I am really not entirely sure how to help you here. It looks like your stuff is working well on the training data at least. I would suggest to look at other issues to see how people have dealt with their train model. I am sorry, I don't think it can help you much here. Can I suggest you to use a LLM to help you craft your questions so they can be a little bit more specific for me? You just did a big dump of information without clear questions you want me to explain.

czy1998916, 2025-04-02:
Sorry, I saw explanations for other parameters, but my main question is how to obtain the dimensions parameter in config.pose. yaml, and what are the issues with my inference results

czy1998916, 2025-04-09:
@TontonTremblay

czy1998916, 2025-04-11:
The problem is that some inference results are good, while others are poor. What is the reason for this. Is it because I arbitrarily set dimensions that affect it

TontonTremblay, 2025-04-11:
welcome to deep learning based computer vision.

TontonTremblay, 2025-04-11:
normal behaviour I think your bounding box dimensions are wrong, you should measure your object.

czy1998916, 2025-04-11:
Does this algorithm not support detecting multiple identical objects

TontonTremblay, 2025-04-11:
If you trained it to do so, it should yes.

czy1998916, 2025-04-13:
Dear author, hello. Below are the inference results of my model trained on 20000 images. Some test results are not very good, I would like to ask what is the optimal number of datasets.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

czy1998916, 2025-04-13:
If I want to use a realsense camera to detect objects in real time and output poses, which inference script should I use? I have found that there are many inference scripts in the project. @TontonTremblay

czy1998916, 2025-04-19:
After training the model multiple times, I discovered another issue. When I train with -- objects=['Maid '] to specify an object of interest, the terminal will output NaN or Inf found in input tensor. and the loss function will explode. My terminal output is as follows
(NVISII)  rx@rx :~/ 2_Deep_Object_Pose/scripts/train2$ python -m torch.distributed.run --nproc_per_node=1 train.py --local_rank=0 --data /home/rx/Deep_Object_Pose/data_generation/nvisii_data_gen/output/output_example
[INFO] 2025-04-19 15:39:58,055 run: Running torch.distributed.run with args: ['/home/rx/anaconda3/envs/NVISII/lib/python3.9/site-packages/torch/distributed/run.py', '--nproc_per_node=1', 'train.py', '--local_rank=0', '--data', '/home/rx/Deep_Object_Pose/data_generation/nvisii_data_gen/output/output_example']
[INFO] 2025-04-19 15:39:58,056 run: Using nproc_per_node=1.
[INFO] 2025-04-19 15:39:58,056 api: Starting elastic_operator with launch configs:
entrypoint       : train.py
min_nodes        : 1
max_nodes        : 1
nproc_per_node   : 1
run_id           : none
rdzv_backend     : static
rdzv_endpoint    : 127.0.0.1:29500
rdzv_configs     : {'rank': 0, 'timeout': 900}
max_restarts     : 3
monitor_interval : 5
log_dir          : None
metrics_cfg      : {}

[INFO] 2025-04-19 15:39:58,056 local_elastic_agent: log directory set to: /tmp/torchelastic_k5_5cgxa/none_kbi2b3zn
[INFO] 2025-04-19 15:39:58,056 api: [default] starting workers for entrypoint: python
[INFO] 2025-04-19 15:39:58,056 api: [default] Rendezvous'ing worker group
[INFO] 2025-04-19 15:39:58,056 static_tcp_rendezvous: Creating TCPStore as the c10d::Store implementation
/home/rx/anaconda3/envs/NVISII/lib/python3.9/site-packages/torch/distributed/elastic/utils/store.py:52:  FutureWarning: This is an experimental API and will be changed in future.
warnings.warn(
[INFO] 2025-04-19 15:39:58,060 api: [default] Rendezvous complete for workers. Result:
restart_count=0
master_addr=127.0.0.1
master_port=29500
group_rank=0
group_world_size=1
local_ranks=[0]
role_ranks=[0]
global_ranks=[0]
role_world_sizes=[1]
global_world_sizes=[1]

[INFO] 2025-04-19 15:39:58,060 api: [default] Starting worker group
[INFO] 2025-04-19 15:39:58,060 __init__: Setting worker0 reply file to: /tmp/torchelastic_k5_5cgxa/none_kbi2b3zn/attempt_0/0/error.json
start: 15:39:58.628429
load data: ['/home/rx/Deep_Object_Pose/data_generation/nvisii_data_gen/output/output_example']
load data: 
training data: 2500 batches
load models
ready to train!
Train Epoch: 1 [0/79989 (0%)]   Loss: 0.036854743957520
Train Epoch: 1 [3200/79989 (4%)]        Loss: 5973.501464843750000
Train Epoch: 1 [6400/79989 (8%)]        Loss: 0.014067125506699

@TontonTremblay @mintar

czy1998916, 2025-04-19:
`(NVISII) rx@rx:~/2_Deep_Object_Pose/scripts/train2$ python -m torch.distributed.run --nproc_per_node=1 train.py --local_rank=0  --data /home/rx/Deep_Object_Pose/data_generation/nvisii_data_gen/output/output_example
[INFO] 2025-04-19 16:18:02,816 run: Running torch.distributed.run with args: ['/home/rx/anaconda3/envs/NVISII/lib/python3.9/site-packages/torch/distributed/run.py', '--nproc_per_node=1', 'train.py', '--local_rank=0', '--data', '/home/rx/Deep_Object_Pose/data_generation/nvisii_data_gen/output/output_example']
[INFO] 2025-04-19 16:18:02,817 run: Using nproc_per_node=1.
[INFO] 2025-04-19 16:18:02,817 api: Starting elastic_operator with launch configs:
  entrypoint       : train.py
  min_nodes        : 1
  max_nodes        : 1
  nproc_per_node   : 1
  run_id           : none
  rdzv_backend     : static
  rdzv_endpoint    : 127.0.0.1:29500
  rdzv_configs     : {'rank': 0, 'timeout': 900}
  max_restarts     : 3
  monitor_interval : 5
  log_dir          : None
  metrics_cfg      : {}

[INFO] 2025-04-19 16:18:02,817 local_elastic_agent: log directory set to: /tmp/torchelastic_qz1vh008/none_tpcglp0j
[INFO] 2025-04-19 16:18:02,817 api: [default] starting workers for entrypoint: python
[INFO] 2025-04-19 16:18:02,817 api: [default] Rendezvous'ing worker group
[INFO] 2025-04-19 16:18:02,817 static_tcp_rendezvous: Creating TCPStore as the c10d::Store implementation
/home/rx/anaconda3/envs/NVISII/lib/python3.9/site-packages/torch/distributed/elastic/utils/store.py:52: FutureWarning: This is an experimental API and will be changed in future.
  warnings.warn(
[INFO] 2025-04-19 16:18:02,818 api: [default] Rendezvous complete for workers. Result:
  restart_count=0
  master_addr=127.0.0.1
  master_port=29500
  group_rank=0
  group_world_size=1
  local_ranks=[0]
  role_ranks=[0]
  global_ranks=[0]
  role_world_sizes=[1]
  global_world_sizes=[1]

[INFO] 2025-04-19 16:18:02,819 api: [default] Starting worker group
[INFO] 2025-04-19 16:18:02,819 __init__: Setting worker0 reply file to: /tmp/torchelastic_qz1vh008/none_tpcglp0j/attempt_0/0/error.json
start: 16:18:03.397244
load data: ['/home/rx/Deep_Object_Pose/data_generation/nvisii_data_gen/output/output_example']
load data: 
training data: 2500 batches
load models
ready to train!
Train Epoch: 1 [0/79989 (0%)]   Loss: 0.036703445017338
Train Epoch: 1 [3200/79989 (4%)]        Loss: 0.006743073929101
Train Epoch: 1 [6400/79989 (8%)]        Loss: 0.006489558611065
Train Epoch: 1 [9600/79989 (12%)]       Loss: 0.007365600671619
Train Epoch: 1 [12800/79989 (16%)]      Loss: 0.007147031836212
Train Epoch: 1 [16000/79989 (20%)]      Loss: 0.006739089265466
Train Epoch: 1 [19200/79989 (24%)]      Loss: 0.006625058129430
Train Epoch: 1 [22400/79989 (28%)]      Loss: 0.006619947496802
Train Epoch: 1 [25600/79989 (32%)]      Loss: 0.006703049410135
Train Epoch: 1 [28800/79989 (36%)]      Loss: 0.006157390773296
Train Epoch: 1 [32000/79989 (40%)]      Loss: 0.006845242343843
Train Epoch: 1 [35200/79989 (44%)]      Loss: 0.006777579896152
Train Epoch: 1 [38400/79989 (48%)]      Loss: 0.006546503864229
Train Epoch: 1 [41600/79989 (52%)]      Loss: 0.006607317831367
Train Epoch: 1 [44800/79989 (56%)]      Loss: 0.006645914167166
Train Epoch: 1 [48000/79989 (60%)]      Loss: 0.006415957119316
Train Epoch: 1 [51200/79989 (64%)]      Loss: 0.006156315095723
Train Epoch: 1 [54400/79989 (68%)]      Loss: 0.006448927801102
Train Epoch: 1 [57600/79989 (72%)]      Loss: 0.006268760655075
Train Epoch: 1 [60800/79989 (76%)]      Loss: 0.006432910449803
Train Epoch: 1 [64000/79989 (80%)]      Loss: 0.006313226185739
Train Epoch: 1 [67200/79989 (84%)]      Loss: 0.006537297740579
Train Epoch: 1 [70400/79989 (88%)]      Loss: 0.006065989844501
Train Epoch: 1 [73600/79989 (92%)]      Loss: 0.006392578594387
Train Epoch: 1 [76800/79989 (96%)]      Loss: 0.006095190998167
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[... 156 lines cut ...]
Train Epoch: 6 [41600/79989 (52%)]      Loss: 0.002967811189592
Train Epoch: 6 [44800/79989 (56%)]      Loss: 0.002770419698209
Train Epoch: 6 [48000/79989 (60%)]      Loss: 0.003337596543133
Train Epoch: 6 [51200/79989 (64%)]      Loss: 0.003102514427155
Train Epoch: 6 [54400/79989 (68%)]      Loss: 0.003383880946785
Train Epoch: 6 [57600/79989 (72%)]      Loss: 0.002825342118740
Train Epoch: 6 [60800/79989 (76%)]      Loss: 0.003313282504678
Train Epoch: 6 [64000/79989 (80%)]      Loss: 0.003276488743722
Train Epoch: 6 [67200/79989 (84%)]      Loss: 0.003480429295450
Train Epoch: 6 [70400/79989 (88%)]      Loss: 0.002859102794901
Train Epoch: 6 [73600/79989 (92%)]      Loss: 0.002977068070322
Train Epoch: 6 [76800/79989 (96%)]      Loss: 0.002983541227877
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
[W pthreadpool-cpp.cc:90] Warning: Leaking Caffe2 thread-pool after fork. (function pthreadpool)
WARNING:tensorboardX.x2num:NaN or Inf found in input tensor.
Train Epoch: 7 [0/79989 (0%)]   Loss: 0.002810227684677
Train Epoch: 7 [3200/79989 (4%)]        Loss: 0.003018322866410
Train Epoch: 7 [6400/79989 (8%)]        Loss: 0.002933786017820
Train Epoch: 7 [9600/79989 (12%)]       Loss: 0.003066249424592`

czy1998916, 2025-04-22:
topic_camera: "/camera/camera/color/image_raw"
topic_camera_info: "/camera/camera/color/camera_info"
topic_publishing: "dope"
input_is_rectified: True   # Whether the input image is rectified (strongly suggested!)
downscale_height: 400      # if the input image is larger than this, scale it down to this pixel height

# Comment any of these lines to prevent detection / pose estimation of that object
weights: {
    # "cracker":"package://dope/weights/cracker_60.pth",
    # "gelatin":"package://dope/weights/gelatin_60.pth",
    # "meat":"package://dope/weights/meat_20.pth",
    # "mustard":"package://dope/weights/mustard_60.pth",
    # "soup":"package://dope/weights/soup_60.pth",
    # 'peg_hole': "package://dope/weights/peg_box_40.pth",
    # 'cube_red': "package://dope/weights/red_40.pth",
    #"sugar":"package://dope/weights/sugar_60.pth"
    # "bleach":"package://dope/weights/bleach_28_dr.pth"
    # 'pudding':"weights_dope/pudding.pth"
    # 'pudding':"weights_dope/dope_network/net_epoch_60.pth"
    #'alphabet_soup':"weights_dope/resnet_simple/alphabe_soup.pth"
    # 'toast_bread': "/home/rx/Deep_Object_Pose/train2/train_tmp/net_epoch_100.pth"
    'Minute_Maid':"/home/rx/2_Deep_Object_Pose/scripts/train2/train_tmp/net_epoch_100.pth"
}

# Type of neural network architecture
architectures: {
    # 'pudding':"dope",
    'Minute_Maid':'dope',   
}

# Cuboid dimension in cm x,y,z
dimensions: {
    # "cracker": [16.403600692749023,21.343700408935547,7.179999828338623],
    # "gelatin": [8.918299674987793, 7.311500072479248, 2.9983000755310059],
    # "meat": [10.164673805236816,8.3542995452880859,5.7600898742675781],
    # "mustard": [9.6024150848388672,19.130100250244141,5.824894905090332],
    # "soup": [6.7659378051757813,10.185500144958496,6.771425724029541],
    # "sugar": [9.267730712890625,17.625339508056641,4.5134143829345703],
    # "bleach": [10.267730712890625,26.625339508056641,7.5134143829345703],
    # "peg_hole": [12.6,3.9,12.6],
    # 'cube_red':[5,5,5],
    # 'pudding':[49.47199821472168, 29.923000335693359, 83.498001098632812],
    # 'toast_bread':[10.4741957, 8.4758271030178, 5.9598632]
    'Minute_Maid':[6.2201642, 16.75548, 6.5692321]
}

class_ids: {
    # "cracker": 1,
    # "gelatin": 2,
    # "meat":    3,
    # "mustard": 4,
    # "soup":    5,
    # "sugar":   6,
    # "bleach":  7,
    # "peg_hole": 8,
    # "cube_red": 9,
    # 'pudding': 10,
    # 'toast_bread': 0,
    'Minute_Maid': 0
}

draw_colors: {
    # "cracker": [13, 255, 128],  # green
    # "gelatin": [255, 255, 255],  # while
    #"toast_bread": [0, 104, 255],  # blue
    # "mustard": [217,12, 232],  # magenta
    # "soup": [255, 101, 0],  # orange
    # "sugar": [232, 222, 12],  # yellow
    # "bleach": [232, 222, 12],  # yellow
    # "peg_hole": [232, 222, 12],  # yellow
    # "toast_bread": [255,0,0],
    # "pudding": [255,0,Minute_Maid
    'Minute_Maid': [255,0,0]
}

# optional: provide a transform that is applied to the pose returned by DOPE
model_transforms: {
#    "cracker": [[ 0,  0,  1,  0],
#                [ 0, -1,  0,  0],
#                [ 1,  0,  0,  0],
#                [ 0,  0,  0,  1]]
}

# optional: if you provide a mesh of the object here, a mesh marker will be
# published for visualization in RViz
# You can use the nvdu_ycb tool to download the meshes: https://github.com/NVIDIA/Dataset_Utilities#nvdu_ycb
meshes: {
#    "cracker": "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/003_cracker_box/google_16k/textured.obj",
#    "gelatin": "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/009_gelatin_box/google_16k/textured.obj",
#    "meat":    "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/010_potted_meat_can/google_16k/textured.obj",
#    "mustard": "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/006_mustard_bottle/google_16k/textured.obj",
#    "soup":    "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/005_tomato_soup_can/google_16k/textured.obj",
#    "sugar":   "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/004_sugar_box/google_16k/textured.obj",
#    "bleach":  "file://path/to/Dataset_Utilities/nvdu/data/ycb/aligned_cm/021_bleach_cleanser/google_16k/textured.obj",
    # "toast_bread": "/home/rx/3d_model/obj/toast_bread/toast_bread.obj"
    # "Minute_Maid": "/home/rx/3d_model/obj/Minute_Maid/Minute_Maid.obj"
}

# optional: If the specified meshes are not in meters, provide a scale here (e.g. if the mesh is in centimeters, scale should be 0.01). default scale: 1.0.
mesh_scales: {
    # "cracker": 0.01,
    # "gelatin": 0.01,
    # "meat":    0.01,
    # "mustard": 0.01,
    # "soup":    0.01,
    # "sugar":   0.01,
    # "bleach":  0.01,
    # "toast_bread": 4.0
    "Minute_Maid": 4.0
}

# Config params for DOPE
thresh_angle: 0.5
thresh_map: 0.0001
sigma: 3
thresh_points: 0.1

The above file content makes up my configuration file content.

I have checked the config file and found that the purpose of mesh is to map the obj model of the object of interest to the object, but I did not see any effect when I ran the program. Can you tell me what the problem is?

[image]
[image]
[image]
[image]
[image]
[image]

czy1998916, 2025-04-22:
@TontonTremblay @mintar

czy1998916, 2025-04-28:
The issue has been resolved

closed 2025-05-09 by nv-jeff