NVlabs/Deep_Object_Pose #406: Implementing dope_pose on robots

czy1998916, 2025-05-16:
Dear author, hello. Thank you for your outstanding contribution. Below is a demo of using your algorithm.
May I ask if this algorithm can track objects. Is there an algorithm that can be used to identify multiple objects and has tracking capabilities with a single model.

czy1998916, 2025-05-16:
Sorry, it seems that my demo cannot be uploaded, so I have generated a link
https://vvp4.bmwwx.cn/iNKXw3

TontonTremblay, 2025-05-16:
yeah there are a lot of options, look around.

czy1998916, 2025-05-16:
https://github.com/user-attachments/assets/55c4ae40-c768-49a7-9fb9-081e905cc044、
You should be able to watch it this time. Do you have any recommendations about algorithms @TontonTremblay

czy1998916, 2025-05-16:
https://github.com/user-attachments/assets/4c3a78a4-3053-4ce0-a497-bc7da6543c71

You should be able to watch it this time. Do you have any recommendations about algorithms

Manu752, 2025-05-16:
@ czy1998916 did you use the ros 1 package to implement this ?

czy1998916, 2025-05-16:
If you haven't output any key points, it means you haven't detected the object, so naturally there's no pose information. We suggest you change the scene or perspective

Manu752, 2025-05-19:
ok which topic are you talking about ? . my command is stuck here 

''
process[dope-1]: started with pid [56595]
/usr/local/lib/python3.8/dist-packages/scipy/__init__.py:143: UserWarning: A NumPy version >=1.19.5 and <1.27.0 is required for this version of SciPy (detected version 1.17.4)
  warnings.warn(f"A NumPy version >={np_minversion} and <{np_maxversion}"
Loading DOPE model '/root/catkin_ws/src/dope/weights/Milk.pth'...
/usr/local/lib/python3.8/dist-packages/torchvision/models/_utils.py:208: UserWarning: The parameter 'pretrained' is deprecated since 0.13 and may be removed in the future, please use 'weights' instead.
  warnings.warn(
/usr/local/lib/python3.8/dist-packages/torchvision/models/_utils.py:223: UserWarning: Arguments other than a weight enum or `None` for 'weights' are deprecated since 0.13 and may be removed in the future. The current behavior is equivalent to passing `weights=None`.
  warnings.warn(msg)
/root/catkin_ws/src/dope/src/dope/inference/detector.py:228: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  net.load_state_dict(torch.load(path))
    Model loaded in 2.164874315261841 seconds.
Running DOPE...  (Listening to camera topic: '/dope/webcam/image_raw')
Ctrl-C to stop
''  

nothing happens after it is listening to topic

Manu752, 2025-05-19:
did you retrain it for your items or is it running based on the default weights ?

czy1998916, 2025-05-19:
There is no problem with starting the program. Based on my experience, there are two reasons: 1. Your program was not trained properly. 2. Your detection angle is incorrect. （The fundamental reason is that the model is not well trained）

czy1998916, 2025-05-19:
You can try holding the object further away (I don't know why, sometimes when the object is too close, it just can't be detected)

czy1998916, 2025-05-19:
I don't understand what you mean, please explain it in more detail

Manu752, 2025-05-19:
this is my belif map . i am trying to detect milk carton -

[image]

Manu752, 2025-05-19:
do i need to retrain for milk carton ? can i not use the default weights ?

czy1998916, 2025-05-19:
The effect is very poor. I suggest you change your perspective and face the camera directly to find the most suitable angle

czy1998916, 2025-05-19:
I think you can try multiple perspectives first and then use predictive scripts to test the effectiveness of your model. Then decide whether to retrain. If you retrain, I suggest you use a different dataset

Manu752, 2025-05-19:
Can you share your belief map ?

Manu752, 2025-05-23:
Hi again , can you share which method did you use to train your model ? Did you use blenderproc?

closed 2025-07-17 by czy1998916